FreeOpcUa/python-opcua #1015: read_raw_history(). OSError: [WinError 10038]

leonardowhich, 2020-05-03:
Good evening, I tried to create an example opc ua client for reading historical data. Received the following error. Please tell me where I made a mistake?
```python
from opcua import Client
import os
```

```python
client=Client("opc.tcp://YOKOGAWA:YOKOGAWA@apc0466:34430")
client.set_security_string("Basic256,SignAndEncrypt,certificate-example.der,private-key-example.pem")
client.application_uri='urn:example.org:FreeOpcUa:python-opcua'
```

    DEPRECATED! Do not use SecurityPolicyBasic256 anymore!
    

```python
try:
    client.connect()
    tag1 = client.get_node("ns=4;s=Root/Objects/APC/Applications/CDU_App/Live/Variables/AI133C.PV/MeasuredValueValidated")
    print("tag1 is: {0} [{2}] with value {1} ".format(tag1, tag1.get_value(), tag1.get_description()))    
finally:
    client.disconnect()
result=tag1.read_raw_history(starttime=datetime(2017,5,4,13,44,30), endtime=datetime(2017,5,4,16,59,30), numvalues=10)
```

    tag1 is: Node(StringNodeId(ns=4;s=Root/Objects/APC/Applications/CDU_App/Live/Variables/AI133C.PV/MeasuredValueValidated)) [LocalizedText(Encoding:0, Locale:None, Text:None)] with value nan 
    

    ---------------------------------------------------------------------------

    OSError                                   Traceback (most recent call last)

    <ipython-input-85-a637592e157c> in <module>
          5 finally:
          6     client.disconnect()
    ----> 7 result=tag1.read_raw_history(starttime=datetime(2017,5,4,13,44,30), endtime=datetime(2017,5,4,16,59,30), numvalues=10)
    

    c:\users\lofa\appdata\local\programs\python\python37\lib\site-packages\opcua\common\node.py in read_raw_history(self, starttime, endtime, numvalues)
        506         details.NumValuesPerNode = numvalues
        507         details.ReturnBounds = True
    --> 508         result = self.history_read(details)
        509         return result.HistoryData.DataValues
        510 
    

    c:\users\lofa\appdata\local\programs\python\python37\lib\site-packages\opcua\common\node.py in history_read(self, details)
        523         params.ReleaseContinuationPoints = False
        524         params.NodesToRead.append(valueid)
    --> 525         result = self.server.history_read(params)[0]
        526         return result
        527 
    

    c:\users\lofa\appdata\local\programs\python\python37\lib\site-packages\opcua\client\ua_client.py in history_read(self, params)
        600         request = ua.HistoryReadRequest()
        601         request.Parameters = params
    --> 602         data = self._uasocket.send_request(request)
        603         response = struct_from_binary(ua.HistoryReadResponse, data)
        604         self.logger.debug(response)
    

    c:\users\lofa\appdata\local\programs\python\python37\lib\site-packages\opcua\client\ua_client.py in send_request(self, request, callback, timeout, message_type)
         79         returns response object if no callback is provided
         80         """
    ---> 81         future = self._send_request(request, callback, timeout, message_type)
         82         if not callback:
         83             data = future.result(self.timeout)
    

    c:\users\lofa\appdata\local\programs\python\python37\lib\site-packages\opcua\client\ua_client.py in _send_request(self, request, callback, timeout, message_type)
         70 
         71             msg = self._connection.message_to_binary(binreq, message_type=message_type, request_id=self._request_id)
    ---> 72             self._socket.write(msg)
         73         return future
         74 
    

    c:\users\lofa\appdata\local\programs\python\python37\lib\site-packages\opcua\common\utils.py in write(self, data)
        116 
        117     def write(self, data):
    --> 118         self.socket.sendall(data)
        119 
        120 
    

    OSError: [WinError 10038] Сделана попытка выполнить операцию на объекте, не являющемся сокетом

AndreasHeine, 2020-05-03:
Сделана попытка выполнить операцию на объекте, не являющемся сокетом
for all not russians... like me:
An attempt was made to perform an operation on an object that is not a socket

AndreasHeine, 2020-05-03:
```
try:
    client.connect()
    tag1 = client.get_node("ns=4;s=Root/Objects/APC/Applications/CDU_App/Live/Variables/AI133C.PV/MeasuredValueValidated")
    print("tag1 is: {0} [{2}] with value {1} ".format(tag1, tag1.get_value(), tag1.get_description()))    
finally:
    client.disconnect()
result=tag1.read_raw_history(starttime=datetime(2017,5,4,13,44,30), endtime=datetime(2017,5,4,16,59,30), numvalues=10)
```
 you disconnect an then you read from the node!?

leonardowhich, 2020-05-04:
> ```
> try:
>     client.connect()
>     tag1 = client.get_node("ns=4;s=Root/Objects/APC/Applications/CDU_App/Live/Variables/AI133C.PV/MeasuredValueValidated")
>     print("tag1 is: {0} [{2}] with value {1} [...]

Thank you very much, elementary) in the evening I'm already not thinking clearly))

closed 2020-05-04 by leonardowhich