FreeOpcUa/python-opcua #1017: Connecting Siemens  WinCC Server with Python FreeOPCUA to suscribe for events

rajnishnationfirst, 2020-05-06:
### **Connecting Siemens  WinCC Server with Python FreeOPCUA to suscribe for events** 

Hi everyone,

I am using FreeOPCUA Python SDK for connecting to Siemens WinCC Server and suscribing to  events  .
By default i am suscribing to Server Node & BaseEventType  using 
# All events at "Server"-node:
 **eventSubscription = subscription.subscribe_events().**

Also i am and suscribing through root.get_child(
I am able to connect but i am not able to suscribe to the events  and getting errors as :

`  **# Client has a few methods to get proxy to UA nodes that should always be in address space such 
       as Root or Objects
        root = client.get_root_node()
        print("Objects node is: ", root)

        # Now getting a variable node using its browse path
        obj = root.get_child(["0:Objects","0:Server"])
        print("MyObject is: ", obj)

        myevent = root.get_child(["0:Types", "0:EventTypes", "0:BaseEventType"])
        print("MyFirstEventType is: ", myevent)

       msclt = SubHandler()
        sub = client.create_subscription(1000, msclt)
        handle = sub.subscribe_events(obj, myevent)

        embed()
        sub.unsubscribe(handle)
        sub.delete()**`

Error:
opc.tcp://EC2AMAZ-VJQV23P:4862 opc.tcp
MessageSecurityMode.None_ MessageSecurityMode.None_
http://opcfoundation.org/UA/SecurityPolicy#None http://opcfoundation.org/UA/SecurityPolicy#None
Requested session timeout to be 3600000ms, got 300000ms instead
ServiceFault from server received while waiting for publish response
Traceback (most recent call last):
  File "C:\Users\WinCC\Desktop\Wincc sms project\SDK FILES\Events_Test.py", line 25, in <module>
    eventSubscription = subscription.subscribe_events()
  File "C:\Users\WinCC\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\common\subscription.py", line 212, in subscribe_events
    return self._subscribe(sourcenode, ua.AttributeIds.EventNotifier, evfilter, queuesize=queuesize)
  File "C:\Users\WinCC\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\common\subscription.py", line 230, in _subscribe
    mids[0].check()
  File "C:\Users\WinCC\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadMonitoredItemFilterInvalid: The monitored item filter parameter is not valid.(BadMonitoredItemFilterInvalid) 

Attachements of WinCC View Tree Nodes & Issue of Python OPC UA Client getting from WinCC:

[image]
[image]

**Code Files For Review : 3 Code Files  Tested .**

[client-events_1.txt](https://github.com/FreeOpcUa/python-opcua/files/4585566/client-events_1.txt)
[client-events_2.txt](https://github.com/FreeOpcUa/python-opcua/files/4585567/client-events_2.txt)
[client-events_3.txt](https://github.com/FreeOpcUa/python-opcua/files/4585568/client-events_3.txt)

Request anyonw on how can i suscribe to events on WinCC without errors.

AndreasHeine, 2020-05-06:
Hey,
i tested your code against the event example from python-opcua
[client.py.txt](https://github.com/FreeOpcUa/python-opcua/files/4585762/client.py.txt)
[server.py.txt](https://github.com/FreeOpcUa/python-opcua/files/4585763/server.py.txt)

it works! so the error tells you about a timeout what is the server timeout set to?

rajnishnationfirst, 2020-05-06:
Thanks AndreasHeine , for your reply.

**We have sorted the TimeOut Issue on the server side  and set to 3000000ms to macth the OPCUA Client Python SDK Program in the opcua client.py file .**

**Also could you pls assit me  on what configurations you have done on server side ( configurations ) and client side ( python opcUa SDK Settings in client.py file )**

**Pls check all screenshots .**

**But we are still getting the below error :**

Python 3.8.2 (tags/v3.8.2:7b3ab59, Feb 25 2020, 22:45:29) [MSC v.1916 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license()" for more information.
>>> 
== RESTART: C:\Users\WinCC\Desktop\Wincc sms project\SDK FILES\Events_Test.py ==
opc.tcp://EC2AMAZ-VJQV23P:4862 opc.tcp
MessageSecurityMode.None_ MessageSecurityMode.None_
http://opcfoundation.org/UA/SecurityPolicy#None http://opcfoundation.org/UA/SecurityPolicy#None
**ServiceFault from server received while waiting for publish response**
Traceback (most recent call last):
  File "C:\Users\WinCC\Desktop\Wincc sms project\SDK FILES\Events_Test.py", line 25, in <module>
    eventSubscription = subscription.subscribe_events()
  File "C:\Users\WinCC\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\common\subscription.py", line 212, in subscribe_events
    return self._subscribe(sourcenode, ua.AttributeIds.EventNotifier, evfilter, queuesize=queuesize)
  File "C:\Users\WinCC\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\common\subscription.py", line 230, in _subscribe
    mids[0].check()
  File "C:\Users\WinCC\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadMonitoredItemFilterInvalid: The monitored item filter parameter is not valid.(BadMonitoredItemFilterInvalid)
>>> 

Also pls check the attachement of the Server Time Out Configuration File  / Python Error Console & the opcua client.py file settings.

[image]
[image]
[image]
[image]

Now i am only getting the error : 
**ServiceFault from server received while waiting for publish response**

AndreasHeine  , pls check the below code files on which file willbe running ok without this errors.

Pls find the below attached files of codes : All 3 clients 

[client-events_1.txt](https://github.com/FreeOpcUa/python-opcua/files/4586338/client-events_1.txt)
[client-events_2.txt](https://github.com/FreeOpcUa/python-opcua/files/4586339/client-events_2.txt)
[client-events_3.txt](https://github.com/FreeOpcUa/python-opcua/files/4586340/client-events_3.txt)

Thanks , AndreasHeine will wait for your reply.

rajnishnationfirst, 2020-05-06:
Can anyone help me for my post 👍 
https://github.com/FreeOpcUa/python-opcua/issues/1017

Anyone can help me out.?

ts4iot, 2022-12-22:
Hi [rajnishnationfirst](https://github.com/rajnishnationfirst),

I'm facing the same problem. Did you get it run?