FreeOpcUa/python-opcua #1018: Extension object as a result of get_data_value()

marekvermes, 2020-05-06:
Hi!

I'm trying to read data from OPC UA server using get_data_value(), but the result is Extension object. In UA Expert I can see data, it's FloatArray[2048]. How can I "unpack" this data? 
Thanks for any advice!

[image]

[image]

AndreasHeine, 2020-05-06:
have you try "get_value()" it should return a list!?

AndreasHeine, 2020-05-06:
server.py:
```
try:
    from opcua import ua, Server
    from time import sleep
except ImportError as e:
    print(e)

if __name__ == "__main__":
    """
    OPC-UA-Server Setup
    """
    server = Server()

    endpoint = "opc.tcp://127.0.0.1:4840"
    server.set_endpoint(endpoint)

    servername = "Python-OPC-UA"
    server.set_server_name(servername)
    address_space = server.register_namespace(servername + "/namespace")
    
    uri = "urn:opcua:python:server"
    server.set_application_uri(uri)

    server.set_security_policy([ua.SecurityPolicyType.NoSecurity])

    """
    OPC-UA-Modeling
    """
    root_node = server.get_root_node()
    object_node = server.get_objects_node()
    server_node = server.get_server_node()

    parameter_obj = server.nodes.objects.add_object(address_space, "Parameter")
    random_node = parameter_obj.add_variable(address_space, "random", ua.Variant([10,0,20,0,2,0,0,30,1], ua.VariantType.UInt32))

    """
    OPC-UA-Server Start
    """
    server.start()

    try:
        while 1:
            sleep(1)
    except KeyboardInterrupt:
        server.stop()
```

client.py:
```
from opcua import Client, ua

url = "opc.tcp://127.0.0.1:4840"
client = Client(url)

client.connect()
print(f"Connected to: {url}")

node = client.get_node("ns=2;i=2")

node_datavalue = node.get_data_value()
print(node_datavalue)

node_value = node.get_value()
print(node_value)

for node in node_value:
    print(node)

client.disconnect()
```

output:
```
Connected to: opc.tcp://127.0.0.1:4840
DataValue(Value:Variant(val:[10, 0, 20, 0, 2, 0, 0, 30, 1],type:VariantType.UInt32), StatusCode:StatusCode(Good), SourceTimestamp:2020-05-06 09:25:17.680370)
[10, 0, 20, 0, 2, 0, 0, 30, 1]
10
0
20
0
2
0
0
30
1
```

marekvermes, 2020-05-06:
yes I tried also get_value() and the result is:
`ExtensionObject(TypeId:FourByteNodeId(ns=2;i=5001), Encoding:1, 127 bytes)`

AndreasHeine, 2020-05-06:
is it a custom type?

marekvermes, 2020-05-06:
sorry, but what do you mean by a custom type?

AndreasHeine, 2020-05-06:
could you provide a screenshot from the node in ua expert not from edit value?

marekvermes, 2020-05-06:
[image]

AndreasHeine, 2020-05-06:
ok its a custom type ("DynamicDataType")
```
client.connect()
client.load_type_definitions()
```

and test it again pls!

marekvermes, 2020-05-06:
So the results are below. BTW, the update rate of server is 10 minutes. So every 10 min I see new data in UA Expert. I was thinking that maybe I need to make subscription for the node. 
**for get_value():**
`DynamicDataType(MeasurementId: 00000000-0000-0000-0000-000000000000, UTCTimestamp: 1601-01-01 00:00:00, SampleTypeID: 0, Data: [], NumberOfSamples: 0, UnitName: mil, Unit: 5059635, SubunitName: , Subunit: 0, DataStatus: 0, NodeStatus: 0, RPM: 0.0, Fmax: 0.0, FmaxUnitName: Hz, FmaxUnit: 4740186, SamplingPeriod: 0.0, SamplingPeriodUnitName: s, SamplingPeriodUnit: 5457219, ACCoupled: False, DCOffset: 0.0, AlgorithmTypeID: 0, KPHitIndices: [], RevolutionSpeeds: [], PistonAngle: 0.0, PairedMeasurementIDs: [])`

**for get_data_value():**
`DataValue(Value:Variant(val:DynamicDataType(MeasurementId: 00000000-0000-0000-0000-000000000000, UTCTimestamp: 1601-01-01 00:00:00, SampleTypeID: 0, Data: [], NumberOfSamples: 0, UnitName: mil, Unit: 5059635, SubunitName: , Subunit: 0, DataStatus: 0, NodeStatus: 0, RPM: 0.0, Fmax: 0.0, FmaxUnitName: Hz, FmaxUnit: 4740186, SamplingPeriod: 0.0, SamplingPeriodUnitName: s, SamplingPeriodUnit: 5457219, ACCoupled: False, DCOffset: 0.0, AlgorithmTypeID: 0, KPHitIndices: [], RevolutionSpeeds: [], PistonAngle: 0.0, PairedMeasurementIDs: []),type:VariantType.ExtensionObject), StatusCode:StatusCode(Good), SourceTimestamp:2020-05-06 12:53:59.688282)`

AndreasHeine, 2020-05-06:
node_value = node.get_value()
print(type(node_value))

what is the type of node_value?

marekvermes, 2020-05-06:
It's
`<class 'opcua.common.node.Node'>`

AndreasHeine, 2020-05-06:
ok if the result of "get_value()" has a type of "opcua.common.node.Node" then you should be able to apply all method of the node class itself -> incl. get_value()

node_value = node.get_value()
value = node_value.get_value()

looks strange...

marekvermes, 2020-05-06:
This is server setup.. but as far as I can connect I assume that there is nothing wrong.. 
[image]
[image]

AndreasHeine, 2020-05-06:
?

node_value = node.get_value().get_value() does that work? or not!?

marekvermes, 2020-05-06:
it doesn't work.
`value = node_value.get_value()`
`AttributeError: 'DynamicDataType' object has no attribute 'get_value'`

AndreasHeine, 2020-05-06:
[image]

with the first get_value() (red) u get the extention object (<class 'opcua.common.node.Node'>) and with the second get_value (green) you will get propably the value

AndreasHeine, 2020-05-06:
my last idea....

https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/common/node.py

```
    def get_attribute(self, attr):
        """
        Read one attribute of a node
        result code from server is checked and an exception is raised in case of error
        """
        rv = ua.ReadValueId()
        rv.NodeId = self.nodeid
        rv.AttributeId = attr
        params = ua.ReadParameters()
        params.NodesToRead.append(rv)
        result = self.server.read(params)
        result[0].StatusCode.check()
        return result[0]

    def get_attributes(self, attrs):
        """
        Read several attributes of a node
        list of DataValue is returned
        """
        params = ua.ReadParameters()
        for attr in attrs:
            rv = ua.ReadValueId()
            rv.NodeId = self.nodeid
            rv.AttributeId = attr
            params.NodesToRead.append(rv)

        results = self.server.read(params)
        return results
```

zerox1212, 2020-05-06:
What Python IDE are you using? I would use a nice IDE like PyCharm so you can pause execution and inspect the objects. When dealing with extension objects it's critical to inspect what is actually showing up in the node. I don't use extension object much, but it's possible that you are getting a python object at some point. In other words you could access the value by `node.Value.Value.NumberOfSamples` for example, but until you inspect this complex structure it will be hard to trouble shoot with only UA Expert.

oroulet, 2020-05-06:
What is the issue? Yes call load_type_drfinition() then get_value() on your node. Then you get a python object. That object had all the attributes you want

marekvermes, 2020-05-07:
@AndreasHeine thanks for ideas.. if I ask twice for get_value() I get an error:
AttributeError: 'DynamicDataType' object has no attribute 'get_value'

@oroulet the issue is, that in UAExpert I can see FloatArray of 2048 values.. but I can't read this values with get_value() or get_data_value().. I think that the problem is, that this array updates every 10 minutes but in the moment of connection, there is nothing and first data comes after a few minutes.. it looks like it's always approximately at hr:00:sec:... then hr:10:sec:.. then hr:20:sec:.. etc..

oroulet, 2020-05-07:
so your issue is that :
yournode.get_value().Data i an empty list and it should not be empty?

marekvermes, 2020-05-07:
well when I run the code, in first seconds/minutes it should be empty but after next update of values on server it should change and I should read float values as shown in my first comment here..

oroulet, 2020-05-07:
what happens if you use subscription? print the value at every change and send it here, so we will see if we get any events

marekvermes, 2020-05-11:
Hello,
so at the moment of new data are supposed to come, I receive error message:
`ConnectionAbortedError: [WinError 10053] An established connection was aborted by the software in your host machine` 

I'm using virtual machine.

AndreasHeine, 2020-05-11:
hate vm's... check your vm and host networksettings! or run it on host!