FreeOpcUa/python-opcua #1021: instantiate custom object type (server-custom-object.py)

AndreasHeine, 2020-05-07:
Hey,

i am experimenting around with the custom object examples and i found something weird at least for me...

code from:
https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-custom-object.py
```
    # Example 2 - create a new object type and a instance of the new object type
    #-------------------------------------------------------------------------------
    types = server.get_node(ua.ObjectIds.BaseObjectType)
    
    object_type_to_derive_from = server.get_root_node().get_child(["0:Types", 
                                                                   "0:ObjectTypes", 
                                                                   "0:BaseObjectType"])
    mycustomobj_type = types.add_object_type(idx, "MyCustomObjectType")
    mycustomobj_type.add_variable(0, "var_should_be_there_after_instantiate", 1.0) # demonstrates instantiate
    
    myobj = objects.add_object(idx, "MyCustomObjectA", mycustomobj_type.nodeid)
    #-------------------------------------------------------------------------------
```

if i add a new object (custom) like "mycustomobj_type" i would expect that after adding a instance it has all variables from the custom object-type (mycustomobj_type)

[image]

what do i miss?

oroulet, 2020-05-07:
look at tests/tests_common.py and the instantiate test. The support for modelling_rule has been added after that example was written. There is test for that feature but maybe the example is missing something. Can you check and send a PR? thanks

oroulet, 2020-05-07:
also the instantiate method has an option for instantiating optional nodes

AndreasHeine, 2020-05-07:
i will have a look!

AndreasHeine, 2020-05-07:
got it! 

```
    # Example 2 - create a new object type and a instance of the new object type
    #-------------------------------------------------------------------------------
    types = server.get_node(ua.ObjectIds.BaseObjectType)
    
    object_type_to_derive_from = server.get_root_node().get_child(["0:Types", 
                                                                   "0:ObjectTypes", 
                                                                   "0:BaseObjectType"])
    mycustomobj_type = types.add_object_type(idx, "MyCustomObjectType")
    var = mycustomobj_type.add_variable(0, "var_should_be_there_after_instantiate", 1.0) # demonstrates instantiate
    var.set_modelling_rule(True)
    myobj = objects.add_object(idx, "MyCustomObjectA", mycustomobj_type.nodeid)
    #-------------------------------------------------------------------------------
```

PR #1022

zerox1212, 2020-05-14:
If you call `instantiate()` to create the instance of your type does it work without modeling rule?

closed 2020-06-01 by AndreasHeine

AndreasHeine, 2020-06-02:
@zerox1212 
looks like this method returns a list not a node... i would expect a node!

`return [Node(parent.server, nid) for nid in nodeids]`

[image]
[image]