FreeOpcUa/python-opcua #1023: Testing subscription - getting BadSessionClosed

DallasThomas, 2020-05-08:
I am having issues setting up and testing a subscription using a version of the modified myclient code and was hoping someone might be able to help give me into some insight as to what I am doing wrong.

My set-up:

For a server I have a basic Beijer IX Developer test program that is configured as the OPC-UA server.  Currently there are 5 tags/variables I am trying to subscribe too.

In UaExpert everything appears as it should.  Variables and values are found under Root->Objects->Tags.

The modified code I am testing is:

`import sys
import logging
import time
import pprint
sys.path.insert(0, "..")

from opcua import Client
from opcua import ua
from opcua import Subscription

class SubHandler(object):
    # Subscription Handler - testing Subscription on Data Change
    def __init__(self, obj):
        self.obj = obj  
    def datachange_notification(self, node, val, data):
        setattr(self.obj, self.obj.browse_name, data.monitored_item.value.value.value)

class value(object):

    def __setattr__(self, key, value):
        print(key, value)
        super().__setattr__(key, value)
    
    def __init__(self, opcua_server, ua_node):
        self.ua_node = ua_node
        self.browse_name = ua_node.get_browse_name().Name
        self.value = ua_node.get_value() 

        handler = SubHandler(self)
        sub = opcua_server.create_subscription(500, handler)
        handle = sub.subscribe_data_change(self.ua_node)

if __name__ == "__main__":

        client = Client("opc.tcp://192.168.1.161:4840")
        client.connect()

        root = client.get_root_node()
        objNode = client.get_objects_node()

        # Now getting a variable node using its browse path
        myvar = root.get_child(["0:Objects", "2:Tags", "2:Colour"])
        variableObj = value(client, myvar)
        time.sleep(10)

    finally:
        client.disconnect()
`

When I try to run this code - it prints out the information from the the server and then crashes with the bad session after sleeping.

This is the output I get

`
(penv) PS C:\Users\dallas.thomas\Desktop\OPC_UA_Testing> python .\opc.py
opc.tcp://ABLETHA465476:4840 opc.tcp
MessageSecurityMode.None_ MessageSecurityMode.None_
http://opcfoundation.org/UA/SecurityPolicy#None http://opcfoundation.org/UA/SecurityPolicy#None
ua_node Node(StringNodeId(ns=2;s=3))
browse_name Colour
value Blue
Exception calling data change handler
Traceback (most recent call last):
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\common\subscription.py", line 149, in _call_datachange
    self._handler.datachange_notification(data.node, item.Value.Value.Value, event_data)
  File ".\opc.py", line 18, in datachange_notification
    setattr(self.obj, self.obj.browse_name, data.monitored_item.value.value.value)
AttributeError: 'MonitoredItemNotification' object has no attribute 'value'
ServiceFault from server received while waiting for publish response
exception calling callback for <Future at 0x2113e8739c8 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\Users\dallas.thomas\.platformio\python37\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\client\ua_client.py", line 483, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionClosed: The session was closed by the client.(BadSessionClosed)
ServiceFault from server received while waiting for publish response
exception calling callback for <Future at 0x2113e88a408 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\Users\dallas.thomas\.platformio\python37\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\client\ua_client.py", line 483, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionClosed: The session was closed by the client.(BadSessionClosed)
`

I would be very much appreciative of any information anyone might have to help me figure this out.  

Thanks
Dallas

AndreasHeine, 2020-05-08:
coud you tell me when the error appears so i mean immediately or after time?

> BadSessionClosed: The session was closed by the client.(BadSessionClosed) ServiceFault from server received while waiting for publish response exception calling callback for <Future at 0x2113e88a408 

i would guess you close the connection without unsubscribe!?

if i simplify your code it works!

my code:
```
import sys
import logging
import time
import pprint
sys.path.insert(0, "..")

from opcua import Client
from opcua import ua
from opcua import Subscription

class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)

# class SubHandler(object):
#     # Subscription Handler - testing Subscription on Data Change
#     def __init__(self, obj):
#         self.obj = obj  
#     def datachange_notification(self, node, val, data):
#         setattr(self.obj, self.obj.browse_name, data.monitored_item.value.value.value)

# class value(object):

#     def __setattr__(self, key, value):
#         print(key, value)
#         super().__setattr__(key, value)
    
#     def __init__(self, opcua_server, ua_node):
#         self.ua_node = ua_node
#         self.browse_name = ua_node.get_browse_name().Name
#         self.value = ua_node.get_value() 

#         handler = SubHandler(self)
#         sub = opcua_server.create_subscription(500, handler)
#         handle = sub.subscribe_data_change(self.ua_node)

if __name__ == "__main__":

    client = Client("opc.tcp://192.168.1.161:4840")
    # client = Client("opc.tcp://127.0.0.1:4840")
    client.connect()

    root = client.get_root_node()
    objNode = client.get_objects_node()

    # Now getting a variable node using its browse path
    myvar = root.get_child(["0:Objects", "2:Tags", "2:Colour"])
    # myvar = root.get_child(["0:Objects", "2:MyDevices", "2:Device1", "0:Value"])

    handler = SubHandler()
    sub = client.create_subscription(500, handler)
    handle = sub.subscribe_data_change(myvar)

    time.sleep(10)

    client.disconnect()
```

DallasThomas, 2020-05-08:
Hello Andreas,

I tested your simplified code above.  At the beginning of the run, I get 

`
(penv) PS C:\Users\dallas.thomas\Desktop\OPC_UA_Testing> python .\test.py
opc.tcp://ABLETHA465476:4840 opc.tcp
MessageSecurityMode.None_ MessageSecurityMode.None_
http://opcfoundation.org/UA/SecurityPolicy#None http://opcfoundation.org/UA/SecurityPolicy#None
Python: New data change event Node(StringNodeId(ns=2;s=3)) Blue
`

After the sleep(10) I get:

`
ServiceFault from server received while waiting for publish response
exception calling callback for <Future at 0x1c6ed4b8848 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\Users\dallas.thomas\.platformio\python37\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\client\ua_client.py", line 483, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionClosed: The session was closed by the client.(BadSessionClosed)
ServiceFault from server received while waiting for publish response
exception calling callback for <Future at 0x1c6ed4ea8c8 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\Users\dallas.thomas\.platformio\python37\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\client\ua_client.py", line 483, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\dallas.thomas\.platformio\penv\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionClosed: The session was closed by the client.(BadSessionClosed)
`

AndreasHeine, 2020-05-08:
ok lets try:

```
sub.unsubscribe(handle)
sub.delete()
```

before 

`    client.disconnect()`

if this does not help further investigation is necessary...

DallasThomas, 2020-05-08:
No badsession this time.  After the sleep I get 

`
ServiceFault from server received while waiting for publish response

ServiceFault from server received while waiting for publish response
`

And then the program quits.

AndreasHeine, 2020-05-08:
do you get the same error if you use one of the example servers?

DallasThomas, 2020-05-08:
That worked - so it looks like it is server related to the Beijer information.  Will contact them.  Thank-you very much for your time Andreas.

DallasThomas, 2020-05-08:
Andreas,

Mind if I ask you another quick question.  Are there instances where the client stays live indefinitely?

oroulet, 2020-05-08:
Ask them why they are not happy with our client. Must be fine crappy small things

DallasThomas, 2020-05-08:
You do not want to get me started on all the s*** I have had with their code development stuff.  Unfortunately there are not tons of other options. grrrrr

DallasThomas, 2020-05-29:
Hey Guys,

Been a long delay I know.  This Covid thing is making lots of things weird and busy so kind of just getting back to this right now.  Branching off of this trimmed down version - I went back to try my original code I posted above with a few slight modifications.  I am still getting the disconnect error, however am ignoring that as what I eventually will be testing is a live service that does not actually disconnect.

That said.  I had a couple of questions I hope you might have some insight on

1. Tags and Subscriptions

In the example a subscription is set based off a single tag

`myvar = root.get_child(["0:Objects", "2:Tags", "2:Dat"])`

If I am adding other Tags would I create a variable for each one and hence a handle for each?  Or would using an array or hash table with all the tags in a single subscription be recommended

2. Using the Tags

Currently in the initial example - tags are printed upon change through:

`def __setattr__(self, key, value):
        print(key, value)
        super().__setattr__(key, value)`

If for monitoring purposes I wanted to have these values update on a webpage or spreadsheet - would the api code be implemented in the __setattr__ instead of the print?

Sorry these might be rather trivial questions - I totally get it.

AndreasHeine, 2020-05-29:
if you want to push them to a webpage use a websocket!

DallasThomas, 2020-05-29:
Thanks Andreas,

Yeah for a webpage a websocket would probably be the best way.  Thanks for the quick response.

DallasThomas, 2020-05-29:
For other purposes without the socket would the setattr function be the best place to implement any further actions upon change?

AndreasHeine, 2020-05-29:
what do you mean? or what do you want to get done?

DallasThomas, 2020-05-29:
Sorry for not being very clear there.  

In this example:

```
 class value(object):

     def __setattr__(self, key, value):
         print(key, value)
         super().__setattr__(key, value)
```

upon data change it is reflected here with the print command, If I wanted to do more with the tag for reflection in a database or other setup is this the best place for implemenation

AndreasHeine, 2020-05-29:
i would think in a different way:
in node.js i work alot with events... just an idea...
```
from eventemitter import EventEmitter
emitter = EventEmitter()

...

class SubHandler(object):
    def datachange_notification(self, node, val, data):
          global emitter
          emitter.emit(str(node) ,val)
```

DallasThomas, 2020-05-29:
Thanks will look into this and give it a try.  Very helpful.

AndreasHeine, 2020-05-30:
i was a little bored...

https://github.com/AndreasHeine/opcua-sub-to-websocket

 #coronamadness ;)

DallasThomas, 2020-06-01:
That is sweet.  I get the coronamadness for sure - mine pulls me in too many directions - kids help with that one too

DallasThomas, 2020-06-04:
Andreas,  

That code is pretty cool.  I have been playing around with it and have been trying some different things with it.  I wondered what would happen if I wanted to drop the websocket and just write the data into a database. (using your opcua-sub-to-websocket code)

I think I might be missing something on just how the code functions because I am having more difficulties with this minor modification than I should.

For testing sake I dropped the events at this point and for nodes I have 5 nodes:

```
nodes_to_subscribe = ["ns=2;s=100", "ns=2;s=101", "ns=2;s=102", "ns=2;s=103", "ns=2;s=104"]
```

Working off the idea I should have all updated nodes prior to db write - i am using a dict to store node values on data_change

I first tried adding this to:

```  
def datachange_notification(self, node, val, data):
        self.node = node
        self.val = val
        dict[node] = val
        #print(node, val, data)
```
I was having difficulties though figuring out from here then how to transition to the connecting and writing to the DB.

I wondered if the ws_handler function was the key - and this is where I discovered my understanding of asyncio needs some work - because my attempt is throwing a typeerror as my asyncio.Future is missing a couroutine...

```
async def ws_handler():
   while 1:
      await asyncio.sleep(0)
      if handler.node:
         dict[handler.node.nodeid] = handler.val
         await db_update()

async def db_update():
   if len(dict) > 4:
      pprint(dict)

asyncio.ensure_future(ws_handler, loop=loop)
```

Would replacing the websocket service with the sql connection be the way to go about this?  Once again thanks for any help you might be able to provide.

AndreasHeine, 2020-06-04:
Thanks! This was just a small "prove of concept" for slow processes on small scale ;)

I know there are situations where data could potentialy missed if a message arrives while websocket.send() hasnot finished its a mix of sync and async code. i am working on a version with asyncua (has better performance) with an message-queue/list kind of thing, which should do the trick.

i am still reading the docs, because i am not sure if a lock is required or pythons gil kicks in...

datachange_notification_queue = []
def datachange_notification(self, node, val, data):
        datachange_notification_queue.append((node, val, data))

if you plan it to use sql, you will need a queue (maybe a list or a sqllite inmemmory database) for sure, sometimes you need to reconnect or wait because of slow network speed

DallasThomas, 2020-06-04:
Yeah trying with a queue well in my case the dict.  What I am finding is - I want to write to the db once all of the nodes have changed.  

so in 

def datachange_notification (self, node, val, data)

I set

dict[node] = val
if len(dict) > 4
  pprint(dict)

which works great for the first one, but then of course the length of dict is always > 4 after this so need to reset that.

If i put 

dict = {} after the pprint - i get a local variable referenced before assignment error.

This part is iterative so not exactly sure why I am getting this.  Any thoughts

AndreasHeine, 2020-06-04:
you need to assign and init outside the subhandler at the very top and if it throws still a error you can try

`global dict`

before using it

```
global dict
dict[node] = val
if len(dict) > 4
    pprint(dict)
    dict = {}
```

but "global" is always a thing ... it should be avoided if possible!

DallasThomas, 2020-06-04:
Yeah - sorry I just realized this - and changed it over to self dict in the init.

I got rid of the error.  It also prints the first set of dict, however does not print anything after even after the nodes have changed :)

DallasThomas, 2020-06-04:
replace dict = {} with dict.clear() and all good

AndreasHeine, 2020-06-04:
development is a sequence of failures with a success at its end ;)

AndreasHeine, 2020-06-05:
@DallasThomas : FYI -> this is what i came up with. https://github.com/AndreasHeine/opcua-sub-to-websocket/blob/master/asyncua%20version/async_client.py

DallasThomas, 2020-06-07:
Very interesting. I was trying to figure out how to pull code tied to a queue out of the data change function of the Subhandler. 

Need to look at this more closely.

DallasThomas, 2020-06-07:
On a side note if instead of a queue you tried a dictionary with node as key. Is there a way to dereference the key such that you could call a specific value using 

Val = dict[“ns=2;s=7”]

and not get a type error

AndreasHeine, 2020-06-07:
tace? i would expect KeyErrors not type errors.

DallasThomas, 2020-06-07:
Testing out your new async client code and have run into something interesting.  When the code first starts up - 

case = 0

In the conditional we get into else of course and set case = 0 then there is the 

await asyncio.sleep(5)

And the program hangs here - it never gets outside this sleep.

DallasThomas, 2020-06-07:
For testing purposes also I have temporarily not included the websocket code but in the notifier have the following

```
async def notifier():
   while 1:
      if users:
         for datachange in sorted(data_notification_dict):
             print("Val: ", data_notification_dict[datachange])
```

Now of course I am expecting something to fail here because nothing is happening with users not having websocket included, etc

DallasThomas, 2020-06-08:
I think it has to do with the changes I did to notifier.  In order to better understand everything I am breaking it into components and testing it a piece at a time.

DallasThomas, 2020-06-09:
Thanks for all your help and suggestions Andre. Very much appreciated.