FreeOpcUa/python-opcua #1024: To list the namespaceUri

yashvanthb, 2020-05-09:
Hi is their any function in client side to list the namespaceUri like 

http://opcfoundation.org/UA/
urn:LAPTOP-0OJQ2QSC:UnifiedAutomation:UaServerCpp
urn:UnifiedAutomation:CppDemoServer:UANodeSetXmlImport

AndreasHeine, 2020-05-09:
get_namespace_array()

yashvanthb, 2020-05-09:
i need in number

AndreasHeine, 2020-05-09:
then number = client.get_namespace_index(uri)

AndreasHeine, 2020-05-09:
please check: https://github.com/FreeOpcUa/python-opcua/tree/master/examples

yashvanthb, 2020-05-09:
thanks for quick replay how to separate the URI using get_namespace_array like if is has 
i am new to opc ua don't mind if asking too many questions

http://opcfoundation.org/UA/
urn:LAPTOP-0OJQ2QSC:UnifiedAutomation:UaServerCpp
urn:UnifiedAutomation:CppDemoServer:UANodeSetXmlImport
like i need in single varibles

AndreasHeine, 2020-05-09:
len(array)

but this is not relatet to python opc ua...

yashvanthb, 2020-05-09:
Thanks AndreasHeine

closed 2020-05-14 by yashvanthb