FreeOpcUa/python-opcua #1025: Connecting WinCC OPCUA Server running on AWS Cloud Ec2 Instance using signing and encryption techniques using free OPCUA Python Client Code &  Basic128rsa15

rajnishnationfirst, 2020-05-11:
Hi , 

**I am trying to connect a **WinCC** **OPCUA** **Server** enabled with Basic128rsa15 with Sign&Encrpyt.

How to generate / get the .der file on / from  WinCC and the .pem file from the WinCC ?

WinCC OPCUA Server is deployed on AWS Windows EC2 Instance.**

**What will be the client connection code that will be used to connect to the WinCC OPC UA Server  enabled with Basic128rsa15 with Sign&Encrpyt. ?**

**Code 1 :** 

```
     client = Client("opc.tcp://localhost:4840/freeopcua/server/")
    #client = Client("opc.tcp://localhost:53530/OPCUA/SimulationServer/")
    client.set_security_string("Basic128rsa15 ,SignAndEncrypt,certificate-example.der,private-key- 
     example.pem")
    client.application_uri = "urn:example.org:FreeOpcUa:python-opcua"
    client.secure_channel_timeout = 10000
    client.session_timeout = 10000
```

Or 

**Code 2:**

   ```
    logging.basicConfig(level=logging.WARN)
    client = Client("opc.tcp://localhost:53530/OPCUA/SimulationServer/")
    client.load_client_certificate("server_cert.pem")
    client.load_private_key("mykey.pem")
 ```
Or

**Code 3:**

```
    logging.basicConfig(level=logging.DEBUG)
    client = Client("opc.tcp://localhost:53530/OPCUA/SimulationServer/")
    client.set_security_string("Basic128rsa15 ,Sign,certificate-example.der,private-key-example.pem")
```

It will be helpful and it will be great anyone from this forum can help and assist  me out .

Good Luck.

AndreasHeine, 2020-05-11:
.der is x509
.pem is x509v3 (v3 = with subject altname) -> thats what you need!

Basic128rsa15 is pretty weak encryption... u should think about a more robust algorithm!

https://github.com/FreeOpcUa/python-opcua/pull/1013

rajnishnationfirst, 2020-05-12:
What is the location of ssl.conf File ?

I cannot locate and change .

Will wait for your reply .

AndreasHeine, 2020-05-12:
it has to be in the location you execute the openssl comands from!

rajnishnationfirst, 2020-05-12:
Thanks Andreas, 
After the .der file and the .pem files generations .

Do we have to keep these client certificates  files on  server side ?

Or also do  we have to export the server certificate to be imported by Python Client.? 

Also there is no forums for .der file and .pem files for Python OPCUA client and Siemens OPCUA Sever. 

If you have exact links or more on detailed it will be very great of you. 

I will wait for your reply. 
Thanks. 

Rajnish Vishwakarma

AndreasHeine, 2020-05-12:
as far as i know you dont need to import/export certificates in sign&encrypt! the thing with more information is that you need do contact siemens customer service if you need deep support you have to buy a siemens support card and you get deeper service/support.

rajnishnationfirst, 2020-05-12:
Thanks Andreas ,

Then if I generate .der file and .pem file for suppose BASE256SHA256 , then how will my sever verify for my self signed generated certificates ?

As given the forum issue case :

https://github.com/FreeOpcUa/python-opcua/issues/978

Thanks 
Rajnish Vishwakarma

AndreasHeine, 2020-05-12:
please this is a topic for siemens! ask them for dokumentation for the wincc opc ua server!