FreeOpcUa/python-opcua #1028: Is their any limitation to free opc ua python

yashvanthb, 2020-05-13:
hi everyone Is there any limitation in this python-opcua 
1. Is there a connection timeout on the client-side/ server-side because [i used C++ SDK their was ] a limitation on client-side and server-side like connection time out (https://www.unified-automation.com/downloads/opc-ua-development.html) 
2. Limitation of on the server connection on the client-side.
if their any limitation other then this please let me know because there is immediate attention for me to develop the clinet

zerox1212, 2020-05-14:
This is a free open source project... there are no intentional limitations.

yashvanthb, 2020-05-14:
Thanks for the reply zerox1212 
 is there any server connection limitation on the client-side.
what should I do if get simple doubts I can't create new issues here

zerox1212, 2020-05-23:
I guess the limitations would be from Python. I have no idea what limitations could arise. Are you trying to have 1000 clients or something?

yashvanthb, 2020-05-28:
for example, there is a situation where  i need to connect to 10-20 different server and access their data, do single client will able to connect to all servers