FreeOpcUa/python-opcua #1029: Release 0.98.10 BadCommunicationError, reason:Chunk size (132) exceeded maximum (-1)

TristanHorlamus, 2020-05-13:
## Error message:
    Connect to opc.tcp://1.2.3.4:4840/OpcUa
    Received an error: MessageAbort(error:StatusCode(BadCommunicationError), reason:Chunk size (132) exceeded maximum (-1))
    Received an error: MessageAbort(error:StatusCode(BadCommunicationError), reason:Chunk size (132) exceeded maximum (-1))

## Code:
    from opcua import Client
    from opcua import ua
    srv_url = "opc.tcp://1.2.3.4:4840/OpcUa"
    srv_usr = "test"
    srv_pwd = "test123"
    print("Connect to %s" % (srv_url))
    srv = Client(srv_url)
    srv.set_user(srv_usr)
    srv.set_password(srv_pwd)
    srv.connect()

oroulet, 2020-05-13:
Ok. Someone changed that value. But your server is probably broken. What server is it?

TristanHorlamus, 2020-05-13:
wouldn't be able to tell. For the moment I locked down my application to install opcua==0.98.9 till the issue gets resolved.
What's the value I am looking for? Maybe I could just change this after creating the client?

oroulet, 2020-05-16:
def __init__(self):
        self.ProtocolVersion = 0
        self.ReceiveBufferSize = 2**32 - 1
        self.SendBufferSize = 2**32 - 1
        self.MaxMessageSize = 0  # No limits
        self.MaxChunkCount = 0  # No limits

try to change things somewhere in uaprotocol_hand.py and let me know what works

TristanHorlamus, 2020-05-16:
I reverted commit 97caf7f28afa0850441c3cc320838c7bb5b730f4
that seems to solve the problem

oroulet, 2020-05-16:
Can you try to find out what is the max value your server accepts? That might help many users

oroulet, 2020-05-16:
Also what server do you have that crashes on that?

TristanHorlamus, 2020-05-18:
2**31 - 1 is the maximum my server accepts.
Server is "SPPA-T3000-OPC_UA_Server" Vendor is "Siemens"

oroulet, 2020-05-18:
Thanks. Let's make that the default

oroulet, 2020-05-18:
Can you send a or?

TristanHorlamus, 2020-05-18:
not sure what 'send or' means

closed 2020-05-18 by TristanHorlamus