FreeOpcUa/python-opcua #1031: Server and Client are dropping the connecction when client call the method from server side

Bannigo, 2020-05-15:
Following is the method on server side. It will take some time to give the response, until the method completes it's task

def go_to_pose_cont(parent, goal_x, goal_y):
    
    x_goal = goal_x.Value
    y_goal = goal_y.Value
    global Pose_Given, pose_degree, pose_y, pose_x
    Pose_Given = True
    print("Got Target")
    K_linear = 0.03
    K_angular = 0.6

    while not rospy.is_shutdown():
        try:
            inc_x = x_goal-pose_x
            inc_y = y_goal-pose_y
            target_distance = abs(math.sqrt(((x_goal - pose_x) ** 2) + ((y_goal - pose_y) ** 2)))
            desired_angle_goal = math.atan2 (inc_y, inc_x) - math.radians(pose_degree)
            angular = desired_angle_goal * K_angular
           
            linear_speed = target_distance * K_linear
            if linear_speed < 0.07:
                linear_speed = 0.07
            Cmd_Vel_msg.linear.x = linear_speed
            Cmd_Vel_msg.angular.z = angular
            movement_pub.publish(Cmd_Vel_msg)
            
            rate.sleep()
            if target_distance < 0.01:
                break

        except rospy.ROSInterruptException:
            Cmd_Vel_msg.linear.x = 0.0
            Cmd_Vel_msg.angular.z = 0.0
            movement_pub.publish(Cmd_Vel_msg)
            Pose_Given = False
            print("Movement Cancelled")
            break

    print("Reached")
    Cmd_Vel_msg.linear.x = 0.0
    Cmd_Vel_msg.angular.z = 0.0
    movement_pub.publish(Cmd_Vel_msg)
    Pose_Given = False
    return True

This is the registration of method on server:

    goalMethod = Methods.add_method(idx, "pose_to_reach", go_to_pose_cont, [ua.VariantType.Float, ua.VariantType.Float], [ua.VariantType.Boolean])

This is the call from client side
goal_resp = obj.call_method("{}:pose_to_reach".format(idx), 5.0, 5.0)
        print(goal_resp)

This is the error:
Traceback (most recent call last):
  File "/home/bannigo/ifam_ws/src/mybot/opcua_communication/scripts/opcua_client_cmd_vel.py", line 51, in <module>
    client.disconnect()
  File "/usr/lib/python2.7/dist-packages/opcua/client/client.py", line 254, in disconnect
    self.close_session()
  File "/usr/lib/python2.7/dist-packages/opcua/client/client.py", line 471, in close_session
    return self.uaclient.close_session(True)
  File "/usr/lib/python2.7/dist-packages/opcua/client/ua_client.py", line 258, in close_session
    data = self._uasocket.send_request(request)
  File "/usr/lib/python2.7/dist-packages/opcua/client/ua_client.py", line 76, in send_request
    data = future.result(self.timeout)
  File "/usr/lib/python2.7/dist-packages/concurrent/futures/_base.py", line 464, in result
    raise TimeoutError()

oroulet, 2020-05-15:
Methods are supposed to return immediately ( this is be very good practice in networking anyway). If you really want to have long networking calls then use the timeout arg to Client

Bannigo, 2020-05-15:
Hi Sir, Thank you so much for your response. I am very new at Opc ua networking, Can you suggest me where to use timeout args in the client's call method