FreeOpcUa/python-opcua #1033: Secure connection without certificates

dave-cz, 2020-05-20:
I want to connect to OPC-UA server, admin gave me URL, username, password, security_mode=SignAndEncrypt, encryption=Basic256Sha256. He says there are no certificates.

Before admin disable connection without security my client can connect to server and read/write data:

```
client = Client('opc.tcp://xxx:10000/RelianceOpcUaServer')
client.set_user('user')
client.set_password('pass')
client.connect()
opc.tcp://xxx:10000/RelianceOpcUaServer opc.tcp
MessageSecurityMode.SignAndEncrypt MessageSecurityMode.None_
http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 http://opcfoundation.org/UA/SecurityPolicy#None
opc.tcp://xxx:10000/RelianceOpcUaServer opc.tcp
MessageSecurityMode.None_ MessageSecurityMode.None_
http://opcfoundation.org/UA/SecurityPolicy#None http://opcfoundation.org/UA/SecurityPolicy#None
```

but now I get this error

```
ServiceFault from server received  in response to CreateSessionRequest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/user/.local/lib/python3.7/site-packages/opcua/client/client.py", line 279, in connect
    self.create_session()
  File "/home/user/.local/lib/python3.7/site-packages/opcua/client/client.py", line 382, in create_session
    response = self.uaclient.create_session(params)
  File "/home/user/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 278, in create_session
    data = self._uasocket.send_request(request)
  File "/home/user/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 84, in send_request
    self.check_answer(data, " in response to " + request.__class__.__name__)
  File "/home/user/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "/home/user/.local/lib/python3.7/site-packages/opcua/ua/uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSecurityPolicyRejected: The security policy does not meet the requirements set by the server.(BadSecurityPolicyRejected)
```

I found only this example: https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-with-encryption.py but `client.set_security_string()` without certs always raise exception.

It's possible to set security_mode and encryption without certificates?

AndreasHeine, 2020-05-20:
you cant sign anything without a cert and you cant encrypt anything without a key!

dave-cz, 2020-05-21:
@AndreasHeine yes, you're right. After a little investigation I found that my own self-signed certs. are OK and there is working command to generate them:

```
openssl req -x509 -sha256 -nodes -days 3650 -newkey rsa:2048 -keyout privateKey.pem -out certificate.crt
openssl x509 -in certificate.crt -outform der -out cert.der
```

closed 2020-05-21 by dave-cz

AndreasHeine, 2020-05-21:
i recommend you to use propper x509v3 certs!