FreeOpcUa/python-opcua #1034: Storing Subscribed Node Value

PauloSilva98, 2020-05-20:
I know this is probably a very begginer-ish question but I have no clue what I should do get the value of the node I am subscribed to. From searching already opened issues I know I have te define something in my Subscription handler but not having a lot of experience with object-oriented programming I am struggling to see the solution.
This is my code: 
```python
    import sys
    sys.path.insert(0, "..")
    import logging
    from time import sleep
    import socket
    import threading
    import xml.etree.ElementTree as ET
    import pymysql

    try:
        from IPython import embed
    except ImportError:
        import code

        def embed():
            vars = globals()
            vars.update(locals())
            shell = code.InteractiveConsole(vars)
            shell.interact()

    from opcua import Client
    from opcua import ua
    from opcua import Subscription

    class SubHandler(object):

        """
        Subscription Handler. To receive events from server for a subscription
        data_change and event methods are called directly from receiving thread.
        Do not do expensive, slow or network operation there. Create another 
        thread if you need to do such a thing
        """

        def datachange_notification(self, node, val, data):
            print("Python: New data change event", node, val)
            setattr(self.obj, self.obj.browse_name, data.monitored_item.Value.Value.Value)

        def event_notification(self, event):
            print("Python: New event", event)

    def atualiza_load_space():

            espaco1 = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.c7t3.espaco")
            espaco2 = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.c7t4.espaco")
            espaco3 = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.c7t5.espaco")

            handler = SubHandler()
            sub = client.create_subscription(100, handler)
            print("Subscription")
            handle_espaco_1 = sub.subscribe_data_change(espaco1)
            handle_espaco_2 = sub.subscribe_data_change(espaco2)
            handle_espaco_3 = sub.subscribe_data_change(espaco3)
        

            handler.

    """
            while(1) :
                if (handler.event_notification(handle_espaco_1)):
                    print("it's lit")
                    #cursor.execute("UPDATE `Pushers` SET `size`= %d WHERE pusher_number = 3" handler.)
                    #mydb.commit()

    """      

    if __name__ == "__main__":

        try:
            UDP_IP = "127.0.0.1" #ip udp 
            UDP_PORT = 54321 #porta udp
            sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM) 
            sock.bind((UDP_IP, UDP_PORT))

            mydb = pymysql.connect("db.fe.up.pt","up201604015", "kHTEYnYP2", "up201604015") #host/user/pass/basedados # SQL connection
            print(mydb) #connected
            cursor=mydb.cursor()
            

            client = Client("opc.tcp://DESKTOP-AFIK9N6:4840")  # OPCUA connection 
            client.connect()
            atualiza_load_space():
            sleep(20)

        finally: 
            print("Connections is Over")
            client.disconnect()
```        

What I am trying to do is, every time there is a notification of a value change in the nodes I am subscribed to, I want to update my database with the value of the nodes.

AndreasHeine, 2020-05-20:
you can try:
```
class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)
        #connect to db
        #write/commit {val} to db
        #disconnect from db

    def event_notification(self, event):
        print("Python: New event", event)
```

PauloSilva98, 2020-05-21:
Gonna try it! Also, would have to create another handler if I would want it to do a different task other than that?

PauloSilva98, 2020-05-21:
Got it to work like this! 

```python 

import sys
sys.path.insert(0, "..")
import logging
from time import sleep
import socket
import threading
import xml.etree.ElementTree as ET
import pymysql

try:
    from IPython import embed
except ImportError:
    import code

    def embed():
        vars = globals()
        vars.update(locals())
        shell = code.InteractiveConsole(vars)
        shell.interact()

from opcua import Client
from opcua import ua
from opcua import Subscription

class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)
        
        #num = int(filter(str.isdigit, node))
        s1 = "Node(StringNodeId(ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.c7t3.espaco))"
        s2 = "Node(StringNodeId(ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.c7t4.espaco))"
        s3 = "Node(StringNodeId(ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.c7t5.espaco))"
        print(str(node))
        if str(node) == s1:
            print("Entrei Caralho")
            cursor.execute("UPDATE `Pushers` SET `size`= %d WHERE pusher_number = 1" %val)
            mydb.commit()
           
        if str(node) == s2: 
            cursor.execute("UPDATE `Pushers` SET `size`= %d WHERE pusher_number = 2" %val)
            mydb.commit()
        if str(node) == s3:
            cursor.execute("UPDATE `Pushers` SET `size`= %d WHERE pusher_number = 3" %val)
            mydb.commit()
    

    def event_notification(self, event):
        print("Python: New event", event)

def atualiza_load_space():

        espaco1 = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.c7t3.espaco")
        espaco2 = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.c7t4.espaco")
        espaco3 = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.c7t5.espaco")

        handler = SubHandler()
        sub = client.create_subscription(100, handler)
        print("Subscription")
        handle_espaco_1 = sub.subscribe_data_change(espaco1)
        handle_espaco_2 = sub.subscribe_data_change(espaco2)
        handle_espaco_3 = sub.subscribe_data_change(espaco3)

        sleep(40)
"""
        while(1) :
            if (handler.event_notification(handle_espaco_1)):
                print("it's lit")
                #cursor.execute("UPDATE `Pushers` SET `size`= %d WHERE pusher_number = 3" handler.)
                #mydb.commit()

"""      

if __name__ == "__main__":

    try: 

        UDP_IP = "127.0.0.1" #ip udp 
        UDP_PORT = 54321 #porta udp
        sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM) 
        sock.bind((UDP_IP, UDP_PORT))

        mydb = pymysql.connect("db.fe.up.pt","up201604015", "kHTEYnYP2", "up201604015") #host/user/pass/basedados # SQL connection
        print(mydb) #connected
        cursor=mydb.cursor()

        client = Client("opc.tcp://DESKTOP-AFIK9N6:4840")  # OPCUA connection 
        client.connect()

        ##Activate Threads

        t_pushers = threading.Thread(target=atualiza_load_space)
        logging.info("Main    : before running thread")
        t_pushers.start()
        print("Threading is Working")
     
        # Setting Variables 
    
        piece_e = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.control.piece_e")
        piece_e.set_value(1,ua.VariantType.UInt16)
        destiny = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.control.info.para[4]")
        # value = destiny.get_value() # get value of node as a DataValue object
        #COLUNA 
        destiny.set_value(1,ua.VariantType.Int16)
        # TIME 
        time = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.control.info.tool_time[1][0]")
        time.set_value(15000,ua.VariantType.Int64)
        time = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.control.info.tool_time[3][0]")
        time.set_value(20000,ua.VariantType.Int64)
    
        # MAQUINA 
        linha = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.control.info.tool_linha[1]")
        linha.set_value(0,ua.VariantType.Int16)
        linha = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.control.info.tool_linha[3]")
        linha.set_value(0,ua.VariantType.Int16)
        
        # FERRAMENTA 
        tool = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.control.info.tool[1][0]")
        tool.set_value(1,ua.VariantType.Int16)
        tool = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.control.info.tool[3][0]")
        tool.set_value(2,ua.VariantType.Int16)

        # Saída 
        linha_saida = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.control.info.linha_saida[1]")
        linha_saida.set_value(1,ua.VariantType.Int16)

        sleep(50)

    finally: 

        print("Connections is Over")
        client.disconnect()
            ```

closed 2020-05-21 by PauloSilva98