FreeOpcUa/python-opcua #1035: Get nodeid of ObjectType from namespace that are not in the UA nodeset

andreafresa, 2020-05-21:
Hello,
I am trying to implement my opcua server with your stack.
My server must be compliant to the Robotics Section of OPC UA, so I need to import:

1. DI NodeSet
2. Robotics NodeSet

Now I am trying to get an element from DI (TopologyElementType).
Here it is the code:

```
from opcua import ua, Server
import sys
sys.path.insert(0, "..")
import time

if __name__ == "__main__":

    # setup our server
    server = Server()
    server.set_endpoint("opc.tcp://127.0.0.1:4840/opcua/server/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://opcua.server.io"
    idx = server.register_namespace(uri)

    # get Objects node, this is where we should put our nodes
    objects = server.get_objects_node()
    
    #import the custom Nodeset2.xml
    server.import_xml("Opc.Ua.Di.NodeSet2.xml")
    server.import_xml("Opc.Ua.Robotics.NodeSet2.xml")
    
    # populating our address space
    topo_nodeid = server.get_root_node().get_child(["0:Types", "0:ObjectTypes", "0:BaseObjectType", "1:TopologyElementType"]).nodeid    
    myobj = objects.add_object(idx, "MyObject", topo_nodeid)
    
    # starting!
    server.start()
    
    try:
        count = 0
        while True:
            #print(server.get_namespace_array)
            time.sleep(1)
            count += 0.1
    finally:
        #close connection, remove subcsriptions, etc
        server.stop()
```

What I get is always the same error:
```  File ".\opc.py", line 27, in <module>
    topo_nodeid = server.get_root_node().get_child(["0:Types", "0:ObjectTypes", "0:BaseObjectType", "1:TopologyElementType"]).nodeid
  File "C:\path\Desktop\python-opcua\opcua\common\node.py", line 471, in get_child
    result.StatusCode.check()
  File "C:\path\Desktop\python-opcua\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNoMatch: "The requested operation has no match to return."(BadNoMatch)
``````
I guess there is some problem with the handling of the namespace.
Do you know how to solve it?

oroulet, 2020-05-21:
If the import works then you are good. The error just means there is an error in your path

andreafresa, 2020-05-21:
If ns are not setted by myself, how can I know which ns is pointing to the DI namespace and which ns is pointing to the Robotics ns?

andreafresa, 2020-05-21:
Ok, I solved it with just a bit of debugging 😄 .
One more question: inside the object that it is created I can see using UAExpert that there are the mandatory and optional items from the robotics section. Is it possible with your stack to delete nodes? I don't want to have everything.
Btw, great work!

AndreasHeine, 2020-05-21:
server object has a lot of methods check them out:

https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/server.py```
    def delete_nodes(self, nodes, recursive=False):
        return delete_nodes(self.iserver.isession, nodes, recursive)

```

closed 2020-05-27 by oroulet