FreeOpcUa/python-opcua #1040: Recover Subscription and Re connection

yashvanthb, 2020-05-26:
if the server disconnected / shutdown how to recover subscription and reconnect to the server  to read the data of the previous nodeID and namespace

oroulet, 2020-05-26:
recovering of subscription and session is not supported. you need to simply reconnect and create new subscriptions

yashvanthb, 2020-05-26:
while creating a new subscription when the server is disconnected, do we need server namespace URIs for mapping to create a new subscription to get the values of the same nodes IDs that is being read before disconnection.
                                                    OR
we can delete the exciting subscription and create a new subscription. 

in this, both cases mapping is needed or not, without mapping can we create subscription

oroulet, 2020-05-27:
I cannot understand... In client we try to delete all subscriptions before leaving, so you should safely be able to recreate any subscriptions you want

swamper123, 2020-05-27:
> while creating a new subscription when the server is disconnected
I don't know if you can even create new subscriptions to a node which can't exist, because there is no server anymore.

But @oroulet is right, that you have to reconnect and recreate them again at the moment and this may be the best practice.

I don't know atm where and what the server is storing when you made a subscription of a node and the client disconnects (server still running). Is there a timeout?

I'm also not sure where and what the Client is saving about the subscriptions (I guess Node, Attribute, and any configs?), but I don't know if this would be a good idea to reuse these subs, because the server may have changed node IDs in the meanwhile/while booting up.

yashvanthb, 2020-05-28:
i am using the OPC UA demo server from the OPC foundation which has a time-out of 1 hour.
i am storing node ID of DA, A&E, HDA in config.in file. 
subscribing to the A&E according to event changes i am getting data once the server time out, i am unable to re-subscribe to name to the same A&E nodes with the same handler.
1. my question is do node id will change while re-subscribe to same A&E 
2. Do we need the URI of the server to re-create the subscription?

After timeout, i am restarting the server while restarting i am unable to unsubscribe.
is there anything to related to namespace URI to save while creating a first subscription that will be used when re-subscription.