FreeOpcUa/python-opcua #1043: Need help using security policy & authentication using certificates

SimranSingh89, 2020-05-29:
My OPCUA server is implemented using the open62541 package. 
I'm trying to automate the test cases using freeopcua in my python framework. 

I connect to opcua server like this -
#Create client connection to opcua server
        client = Client("opc.tcp://opcua1:password1@%s:4840"%(dut.mgmt_ip))
        #opcua.Client.set_security_string('Basic256Sha256,Sign')
        try:
            client.connect()
        except:
            self.error_message = 'FAIL - Connection to OPCUA server failed'
            return False

        client.disconnect()
I have commented out set_security_string because till now my server doesn't support security policy. How security policy will work here when it comes into picture? I can see that we have to provide certificate & private key when calling this function. My server supports username/password authentication right now with None security policy. 
When setting security policy using set_security_string, I should provide which certificate & private key here? Should it be a user certificate ? 
In UaExpert, when you select security policy, one can select either username/password for authentication or certificate/private key. 
I'm confused with security policy & certificate usage in OPCUA. Can anyone explain it & how can it be implemented via freeopcua when using only client class ?

Thanks in advance

AndreasHeine, 2020-05-29:
many servers check the certs more or less depends on the stack!
according to opcua spec. i would use a x509v3 with subject altname (subjectAltName = URI:urn:opcua:python:client,IP: 127.0.0.1) some servers need the client cert as a copy (trusted clients).

like this:
```
from opcua import Client, ua

url = "opc.tcp://127.0.0.1:4840"
client = Client(url)

client.set_user("user1")
client.set_password("pw1")
client.set_security_string("Basic256Sha256,Sign,cert.pem,key.pem")

client.connect()
print(f"Connected to: {url}")

servicelevel_node = client.get_node("ns=0;i=2267")

servicelevel_node_value = servicelevel_node.get_value()
print(servicelevel_node_value)

client.disconnect()
```

SimranSingh89, 2020-05-29:
> many servers check the certs more or less depends on the stack!
> according to opcua spec. i would use a x509v3 with subject altname (subjectAltName = URI:urn:opcua:python:client,IP: 127.0.0.1) some [...]

What I don't understand it that when we give this line -
client.set_security_string("Basic256Sha256,Sign,cert.pem,key.pem")
This is using the cert & key for security policy, not user authentication right ?
Where will we give certificate and key for user authentication if we want to use certificate for authentication ?

AndreasHeine, 2020-05-29:
https://github.com/FreeOpcUa/python-opcua/issues/778

TommyTomson585, 2023-03-22:
Hi folks,

I have a question according to this issue. First of all let me give you some background.

I want to connect to a chipping machine that provides itself an OPC UA Server. 
The manufacturer of the machine requires a certificat of the client, which must be imported by the server. 
The server provides a self-signed certificat that has to be imported by the client. Finally the server has the security mode "sign and encrypt", Basic256ha256 with user authentication in a x509 certificate.

In a first step I tried to establish a connection with the "UAExpert" Client. That runs succesfully.

Now when I try to establish a python client, I get the following error: (see attached file)
 
[image]

If I got the discussion right,

in this line 

await client.set_security(SecurityPolicyBasic256Sha256, certificate=certificatClient, private_key=certificatClientKey,

the user certificat is required?

And this code wants the client certificat

        await client.load_client_certificate(certificatUser)
        await client.load_private_key(certificatUserKey)

correct? Actually when I import the client certificat in the server, no key is necessary.

I hope that someone has a hint or can give me explanaition why the connection doesn't establish. 

Thanks in advance for feedback

TommyTomson585, 2023-03-23:
First of all,

sorry for the wrong post. I'm an absolute rookie in github.

Thank you for your respond nevertheless. 

Finally it works. I used the wrong Server certificate

closed 2023-03-23 by AndreasHeine