FreeOpcUa/python-opcua #1044: Client connection with Basic128Rsa15 and sign & encrypt

yashvanthb, 2020-05-30:
i want to create a client with 
1. security policy "Basic128Rsa15" and MessageSeucurityMode "Sign & encrypt" 
2. Username and password.
3. how this files are generated  "certificate-3072-example.der" "certificate-example.der" "my_cert.der" "my_cert.pem" "private-key-3072-example.pem" "private-key-example.pem" 
4. how can i create certificates of server on clientside.
5 .will these files are generated when we connect the client to the server.

AndreasHeine, 2020-05-30:
1. ok:
`
client.set_security_string("Basic128Rsa15,SignAndEncrypt,certificate.der,key.pem")
`
2. ok:
```
client.set_user("user1")
client.set_password("pw1")
```
3. with open ssl for example!
4. i dont really understand for what reason someone would do that...
5. no

yashvanthb, 2020-05-30:
3. where and how this open SSL created, is this will match with server-side 
4. for more secure security  purpose

AndreasHeine, 2020-05-30:
3.
https://www.openssl.org/
Generate your own x509v3 Certificate
Step 1: Change ssl.conf (subjectAltname, country, organizationName, ...)
ssl.conf:
```
[ req ]
default_bits = 2048
default_md = sha256
distinguished_name = subject
req_extensions = req_ext
x509_extensions = req_ext
string_mask = utf8only
prompt = no
[ req_ext ]
basicConstraints = CA:FALSE
nsCertType = client, server
keyUsage = nonRepudiation, digitalSignature, keyEncipherment, dataEncipherment, keyCertSign
extendedKeyUsage= serverAuth, clientAuth
nsComment = "OpenSSL Generated Certificat"
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid,issuer
subjectAltName = URI:urn:opcua:python:server,IP: 127.0.0.1
[ subject ]
countryName = DE
stateOrProvinceName = HE
localityName = HE
organizationName = AndreasHeine
commonName = PythonOpcUaServer
```
Step 2: openssl genrsa -out key.pem 2048
Step 3: openssl req -x509 -days 365 -new -out certificate.pem -key key.pem -config ssl.conf

4.
"certificate-3072-example.der" "certificate-example.der" "my_cert.der" "my_cert.pem" "private-key-3072-example.pem" "private-key-example.pem" <= these are all selfsigned certificates

if you want to improve security you need a CA (Certificate authority) otherwise you can only verify the subjectAltname extention with URN and IP according to the opc ua spec. (and its still a selfsigned certificate so everyone can generate them...)

yashvanthb, 2020-05-30:
i made the security configuration i am getting below error

  Message=No matching endpoints: 3, http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256

  Source=
  StackTrace:
  File "F:\yashvanth\OPCUA\opcuapython\pythonapplication1\pythonapplication1\clinet.py", line 14, in <module>
    client.set_security_string("Basic256Sha256,SignAndEncrypt,certificate-example.der,private-key-example.pem")

AndreasHeine, 2020-05-30:
1. depend where you are in the comand prompt and where you want to generate the cert and key
2. you could openssl is one free way to generate them
3. looks like client trys to connect to server with a different security policy

yashvanthb, 2020-05-30:
Sorry about i didn't see ,now I changed but still getting error as below 
DEPRECATED! Do not use SecurityPolicyBasic128Rsa15 anymore!
Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security."))
Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security."))
Protocol Error
Traceback (most recent call last):
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 101, in _run
    self._receive()
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 121, in _receive
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 131, in _call_callback
    .format(request_id, self._callbackmap.keys())
opcua.ua.uaerrors._base.UaError: No future object found for request: 0, callbacks in list are dict_keys([1]) 
  Message=exception: no description
  Source=
  StackTrace:
  File "F:\yashvanth\OPCUA\opcuapython\pythonapplication1\pythonapplication1\clinet.py", line 18, in <module>
    client.connect()

yashvanthb, 2020-05-30:
> 1. depend where you are in the comand prompt and where you want to generate the cert and key
> 2. you could openssl is one free way to generate them
i am new this opc ua i don't no how to generate the certificates can give some guidance and some links that will help full for creation of client

AndreasHeine, 2020-05-30:
> DEPRECATED! Do not use SecurityPolicyBasic128Rsa15 anymore!

show me your code cant tell u...

yashvanthb, 2020-05-30:
`from opcua import Client
import time

url = "opc.tcp://LAPTOP-0OJQ2QSC:4840/OPCUA/SimulationServer"
client = Client(url)
client.set_security_string("Basic128Rsa15,SignAndEncrypt,certificate-example.der,private-key-example.pem")
client.application_uri = "urn:LAPTOP-0OJQ2QSC:OPCUA:SimulationServer"
client.secure_channel_timeout = 10000
client.session_timeout = 10000
client.connect()
Server_end_points = client.get_endpoints()
print("Endpoints of server".format(Server_end_points))

while True:
        tag1 = client.get_node("ns=0;i=2255")
        print("Tag value")
        #print(tag1.get_value())

        print("tag1 is: {0} with value {1} ".format(tag1, tag1.get_value()))
        time.sleep(1)
`

yashvanthb, 2020-05-30:
where can i add the certificate files while running the client i am using Visual Studio 2017 for implementing the client

AndreasHeine, 2020-05-30:
its not possible to change the certs wile runtime... you open a secure connection with them if you change them you have to reconnect otherwise you cant decrypt...

yashvanthb, 2020-05-30:
> `from opcua import Client
> import time
>
> url = "opc.tcp://LAPTOP-0OJQ2QSC:4840/OPCUA/SimulationServer"
> client = Client(url) [...]

can give the solution for this

AndreasHeine, 2020-05-30:
your servers security setup is different to the client!

yashvanthb, 2020-05-30:
> its not possible to change the certs wile runtime... you open a secure connection with them if you change them you have to reconnect otherwise you cant decrypt...

not in run time when implementing the client how to create cert files(.der) and .pem i am new to this SSL can you provide any links so that i will help full

yashvanthb, 2020-05-30:
> your servers security setup is different to the client!
what is the solution for this

AndreasHeine, 2020-05-30:
https://www.openssl.org/docs/

yashvanthb, 2020-05-30:
thank you so much 
where can i add the certificate files while running the client i am using Visual Studio 2017 for implementing the client

AndreasHeine, 2020-05-30:
client.set_security_string("Basic128Rsa15,SignAndEncrypt,**certificate.pem**,**key.pem**")

yashvanthb, 2020-05-30:
i am still getting error is it because of server

DEPRECATED! Do not use SecurityPolicyBasic128Rsa15 anymore!
Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security."))
Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security."))
Protocol Error
Traceback (most recent call last):
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 101, in _run
    self._receive()
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 121, in _receive
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 131, in _call_callback
    .format(request_id, self._callbackmap.keys())
opcua.ua.uaerrors._base.UaError: No future object found for request: 0, callbacks in list are dict_keys([1])

but when i used opc ua c++ sdk client to connect i am getting all server endpoints i use prosys OPC UA Simulation
but when i tried in python i am getting above error

AndreasHeine, 2020-05-30:
maybe try one of the example servers

https://github.com/FreeOpcUa/python-opcua/tree/master/examples
https://github.com/AndreasHeine/SecurePythonOpcUaServer

yashvanthb, 2020-05-30:
i tried this above examples before also i got this errors
  Message=No matching endpoints: 2, http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
  Source=
  StackTrace:
  File "F:\yashvanth\OPCUA\opcuapython\pythonapplication1\pythonapplication1\opcuaclinet.py", line 13, in <module>
    client.set_security_string("Basic256Sha256,Sign,certificate.pem,key.pem")

AndreasHeine, 2020-05-30:
does it work with ua expert?

yashvanthb, 2020-05-30:
it works in ua expert without security i think the problem in server only 

  Message=[Errno 10013] error while attempting to bind on address ('192.168.0.11', 4840): an attempt was made to access a socket in a way forbidden by its access permissions
  Source=
  StackTrace:
when i trying to run the server [https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-minimal.py] but i set the IP address as (localhost)192.168.0.11 means my PC is their any other IP address i need to set

which software application you are using for development of python

AndreasHeine, 2020-05-30:
no comercial opc ua simulation...
i test my clients/servers against python-opcua, node-opcua and siemens opcua (real hardware)

AndreasHeine, 2020-05-30:
```
Message=[Errno 10013] error while attempting to bind on address ('192.168.0.11', 4840): an attempt was made to access a socket in a way forbidden by its access permissions
Source=
StackTrace:
when i trying to run the server [https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-minimal.py] but i set the IP address as (localhost)192.168.0.11 means my PC is their any other IP address i need to set
```

it means an application uses this port already so it cant be bound...

yashvanthb, 2020-05-31:
can you suggest the opc server which has Security and policies i tested the python when running i am getting errors.
i have set the security mode Sing&Encrypt and policy basic128Rsa15 in prosys opc ua server 
when i try to connect with server i am getting errors below

DEPRECATED! Do not use SecurityPolicyBasic128Rsa15 anymore!
Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security."))
Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security."))
Protocol Error
Traceback (most recent call last):
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 101, in _run
    self._receive()
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 121, in _receive
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 131, in _call_callback
    .format(request_id, self._callbackmap.keys())
opcua.ua.uaerrors._base.UaError: No future object found for request: 0, callbacks in list are dict_keys([1])

AndreasHeine, 2020-05-31:
https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client_to_prosys_cert.py

yashvanthb, 2020-05-31:
> https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client_to_prosys_cert.py

i cannot find the server_cert.pem

AndreasHeine, 2020-05-31:
generate one for your application! dont use the example certs... check prosys requirements for certs! what kind of cert does prosys accept then generate one and use it! maybe prosys needs the client cert as trusted i dont know proys... yout probelm isn't related to python-opcua!

yashvanthb, 2020-05-31:
there are some certs on prosys sides 
"SimulationServer@LAPTOP-0OJQ2QSC_2048" "SimulationServer@LAPTOP-0OJQ2QSC_2048.pem" "SimulationServerCA" "SimulationServerCA.pem"

i run my own application without security connection is happing but with the security, it's failing can we connect to server with only mode and policy's

AndreasHeine, 2020-05-31:
your client needs its own cert and key otherwise you cant use sign and enrypt!

please have a look on [opcua security](https://opcfoundation.org/wp-content/uploads/2017/11/OPC-UA-Security-Advise-EN.pdf) in general then check prosys requirements and configuration and then setup python-opcua-client with the corresponding setup!

yashvanthb, 2020-05-31:
thanks for the documents 
if we set up own client cert will it connect to server, i have doubts do server will have different certs

AndreasHeine, 2020-05-31:
check prosys documentation! i dont know what they accept... for example siemens opc ua server needs the client cert as a copy on server side (trusted clients) read prosys documentation or ask prosys support!

yashvanthb, 2020-05-31:
1. can we use only security mode and policies and with user name and password to connect to the any server.
2.  is there any API from python to create certs on client-side in C++ theri are some API's line 
Below are the API in C++ OPC UA SDK 

2.1. **initializePkiProviderOpenSSL**  Initialize the PKI provider to use the file based OpenSSL certificate store.
2.2 **loadClientCertificateOpenSSL**  Load client certificate and client private key using the file based OpenSSL certificate store.
2.3. **verifyServerCertificate**  this function verifies if the server certificate is trusted by the client.
2.4. **SessionSecurityInfo**  Save certificate in store.

like above API do we have the API in OPC ua python

AndreasHeine, 2020-05-31:
1. sure
2. no its not part of the opcua spec. ... but you could use os.system() to make openssl commands generate certs

os.system("openssl genrsa -out key.pem 2048")
        os.system("openssl req -x509 -days 365 -new -out cert.pem -key key.pem -config ssl.conf")

yashvanthb, 2020-05-31:
what is the API with security mode and policies that can connect to any server for security mode and policies will be basic125Rsha15 and Sign&enctypt

AndreasHeine, 2020-05-31:
you could write a function witch tryes the different security modes in sequence and returns if its successful... those things you have to do yourself
remember "issue" is not a forum, please stop asking questions not related to this topic... all important infos are provided above!

closed 2020-05-31 by yashvanthb