FreeOpcUa/python-opcua #1045: Source of SourceTimestamp

mstfldmr, 2020-05-30:
Hello,

I am using `data.monitored_item.Value.SourceTimestamp.strftime("%Y-%m-%dT%H:%M:%S.%fZ")` in the `datachange_notification` function. There is a slight difference between what I have and what another client has.

Is SourceTimestamp retrieved from the message coming from the OPC-UA server? Or is it generated in the library when the message is received?

oroulet, 2020-05-30:
This is set by the server. Do not the client

mstfldmr, 2020-05-31:
I subscribe to 500 data points. I observe that Kepware is sending multiple data change notifications with the same source timestamp. Another client on the server subscribes to a few data points and sees the notifications of the same data points with different timestamps.

zerox1212, 2020-05-31:
Kepware isn't a very good UA server, but I also wouldn't expect 2 different clients with different subscriptions to get updated with the same timestamp.

mstfldmr, 2020-06-01:
@zerox1212 I don't have the chance to change it, the production systems are already running with Kepware.

I am subscribed to many data points. May it be sending multiple notifications combined with a common timestamp?

AndreasHeine, 2020-06-01:
i would guess that you got two datachanges in one publish interval that would explain the timestamp