FreeOpcUa/python-opcua #1046: client disconnect API

yashvanthb, 2020-05-31:
got an error when running [https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-minimal.py] 

  Message=The requested operation has no match to return.(BadNoMatch)
  Source=
  StackTrace:
  File "F:\yashvanth\OPCUA\opcuapython\pythonapplication1\pythonapplication1\opcuaclinet.py", line 41, in <module>
    client.disconnect()

AndreasHeine, 2020-05-31:
please more details about the error!

when does it happen?
what client / server?
network-setup?

"BadNoMatch" could indicate that something you want to read/get is not found on server!

yashvanthb, 2020-05-31:
> when does it happen?

when i try to connect the client to prosys server and opc ua server 

> what client / server?

client opcuapython 
server prosys/opc ua client 
when i run client in terminal i got this data and core got stuck and gave error message 
*The requested operation has no match to return.(BadNoMatch*
Objects node is:  Node(TwoByteNodeId(i=84))
Children of root are:  [Node(TwoByteNodeId(i=85)), Node(TwoByteNodeId(i=86)), Node(TwoByteNodeId(i=87))]

AndreasHeine, 2020-05-31:
> "BadNoMatch" could indicate that something you want to read/get is not found on server!

your client requests a node wich does not exist in servers addressspace!

closed 2020-05-31 by yashvanthb