FreeOpcUa/python-opcua #1048: Client auto connect to server after disconnect

Ykulkarni-ops, 2020-05-31:
I have server and client working but i have to add feature to client that if it disconnects ti should reconnect the server automatically within few seconds. 

Can someone please help me this issue?

AndreasHeine, 2020-05-31:
https://github.com/AndreasHeine/opcua-sub-to-websocket/blob/master/opcua_client.py

look at async def main()

check cyclic the service level value if it fails disconnect client and restart like in the example above

jrb0043, 2021-01-11:
@AndreasHeine Hi, can you please reupload that file? I can't manage to reconnect to a server even with a keepalive that I have coded myself.

AndreasHeine, 2021-01-11:
@jrb0043 the repo is still existing! ive just renamed opcua_client.py to client.py

jrb0043, 2021-01-11:
> @jrb0043 the repo is still existing! ive just renamed opcua_client.py to client.py

oh thanks :D