FreeOpcUa/python-opcua #1049: Writing value to server

InterloperX, 2020-05-31:
Hi all, 
Can you please tell me what am i doing wrong. I am trying to write a value to a Siemens S7-1200 device, but i am getting errors all the time.

`if __name__ == "__main__":
        client = Client("opc.tcp://dublek.ddns.net:4840")
        try:
                client.connect()
                print("Connected to Server")   
                node = client.get_node("ns=4;i=90")
                node.set_attribute(ua.AttributeIds.Value, ua.DataValue(10.0))

        finally:
                `client.disconnect()``

Here is the list of errors that i get:
`  File "C:\Users\z003y33f\Downloads\python-opcua-master\VoltBit.py", line 15, in <module>
    node.set_attribute(ua.AttributeIds.Value, ua.DataValue(10.0))
  File "C:\Users\z003y33f\Downloads\python-opcua-master\opcua\common\node.py", line 261, in set_attribute
    result[0].check()
  File "C:\Users\z003y33f\Downloads\python-opcua-master\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadTypeMismatch: "The value supplied for the attribute is not of the same type as the attribute"s `value."(BadTypeMismatch)``

AndreasHeine, 2020-05-31:
what plc datatype you want to write? remember python datatypes and siemens datatypes could be different!

InterloperX, 2020-05-31:
You are right, its working. Thank you.

closed 2020-05-31 by InterloperX

SaralTTT, 2021-09-16:
I can't do it, it says "None" . what should i do I have tried this one
node.set_value(ua.DataValue(ua.Variant(10.0, ua.VariantType.Float)))