FreeOpcUa/python-opcua #1050: Certificate Generation in opc ua client

yashvanth95, 2020-06-01:
1. i found (https://github.com/FreeOpcUa/python-opcua/issues/916) but i didn't get how to create/generate (https://github.com/FreeOpcUa/pythonopcua/blob/master/examples/generate_certificate.sh)
"cert.pem"  "key.pem" 
where to run that generate_certificate.sh files in command prompt?
2. how to create only security,  policy based client without certs.

AndreasHeine, 2020-06-01:
the easy way if openssl is installed:
os.system("openssl genrsa -out key.pem 2048")
os.system("openssl req -x509 -days 365 -new -out cert.pem -key key.pem -config ssl.conf")

yashvanth95, 2020-06-01:
1. how to run the generate_certifiate.sh file in windows after installing in openssl 
2.  os.system() where should I add the generate_certificate.sh

AndreasHeine, 2020-06-01:
1. use google sounds like a quick search
2. no

yashvanth95, 2020-06-01:
thanks, AnderasHeine for the quick response.
1. what is use of generate_certificate.sh in os.system() 
2. os.system("openssl genrsa -out key.pem 2048") in this key.pem will generated when i run the openssl

AndreasHeine, 2020-06-01:
1. i dont know... maybe someone was to lazy to write these to lines in the commandprompt or want to make a cronjob out of it to renew certs in an interval
2. if u run "os.system("openssl genrsa -out key.pem 2048")" it executes the comand in the commandprompt in the backround and generates a key.pem file
3. NoSecurity + Username/Password if u use **Basic&sign** you need something to sign with...

yashvanth95, 2020-06-01:
> 3. NoSecurity + Username/Password if u use **Basic&sign** you need something to sign with...
the basic128Rsa15 and sign&encrypt  this will be the mode and policies + without certs possible?

AndreasHeine, 2020-06-01:
you've got a new account @yashvanthb #1044
same question... **please understand that if you use other then "ua.SecurityPolicyType.NoSecurity" you will need a cert.pem and a key.pem**

yashvanth95, 2020-06-01:
i am new to this i got lot of doubts so i ask this questions
ok i got it now thanks for help AndreasHeine

AndreasHeine, 2020-06-01:
https://documentation.unified-automation.com/uasdkhp/1.0.0/html/_l1_opc_ua_fundamentals.html

yashvanth95, 2020-06-01:
i am new to python is there any documents related to opcuapython how to implement client with security&mode(ex: Basic128Rsa15 &sign&encrypt) + certs 
1. what is the maximum nodeID to read including DA + A&E+ HDA  .

AndreasHeine, 2020-06-01:
there are examples:
https://github.com/FreeOpcUa/python-opcua/tree/master/examples

https://documentation.unified-automation.com/uasdkhp/1.0.0/html/_l2_ua_discovery_connect.html
> Three Message Security Modes are supported:
> 
> None
>     No security is applied. 
> Sign
>     All messages are signed but not encrypted. 
> Sign&Encrypt
>     All messages are signed and encrypted. 
> 
> Note
>     For security reasons, the Security Policies Basic128Rsa15 and None as well as the Message Security Mode None should be deactivated by default.

yashvanth95, 2020-06-01:
i generated key.pem and cert.pem 
after that i run the script i got this error 
SecurityStatus = client.set_security_string("Basic128Rsa15, SignAndEncrypt, F:\Yashvanth\OPCUA\opcuapython\opc_client/cert.pem,F:\Yashvanth\OPCUA\opcuapython\opc_client/key.pem")

  Message= SignAndEncrypt
  Source=
  StackTrace:
  File "F:\yashvanth\OPCUA\opcuapython\pythonapplication1\pythonapplication1\opcuaclinet.py", line 9, in <module>
    SecurityStatus = client.set_security_string("Basic128Rsa15, SignAndEncrypt, F:\Yashvanth\OPCUA\opcuapython\opc_client/cert.pem,F:\Yashvanth\OPCUA\opcuapython\opc_client/key.pem")
that location of .pem files are F:\Yashvanth\OPCUA\opcuapython\opc_client/

yashvanth95, 2020-06-01:
//output 
DEPRECATED! Do not use SecurityPolicyBasic128Rsa15 anymore!
SecurityStatus
 None
Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security."))
Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security."))
Protocol Error
Traceback (most recent call last):
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 101, in _run
    self._receive()
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 121, in _receive
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 131, in _call_callback
    .format(request_id, self._callbackmap.keys())
opcua.ua.uaerrors._base.UaError: No future object found for request: 0, callbacks in list are dict_keys([1])
//Error message
  Message=exception: no description
  Source=
  StackTrace:
  File "F:\yashvanth\OPCUA\opcuapython\pythonapplication1\pythonapplication1\opcuaclinet.py", line 14, in <module>
    client.connect()

### 

i got this error when i set the security and add certs below is the code the location of certs F:\Yashvanth\OPCUA\opcuapython\opc_client\

 SecurityStatus = client.set_security_string("Basic128Rsa15,SignAndEncrypt,F:\Yashvanth\OPCUA\opcuapython\opc_client\cert.pem,F:\Yashvanth\OPCUA\opcuapython\opc_client\key.pem")
    print("SecurityStatus\r\n",SecurityStatus)