FreeOpcUa/python-opcua #1051: Do not use SecurityPolicyBasic128Rsa15 anymore!

yashvanth95, 2020-06-01:
if security is different i also  getting the same errors

//code 
   `     
     uri = "opc.tcp://LAPTOP-0OJQ2QSC:4840/OPCUA/SimulationServer"
     if __name__ == "__main__":
    
    client = Client(uri)

    logging.basicConfig(level=logging.WARNING)
    
    client.set_security_string("Basic128Rsa15,SignAndEncrypt,F:\Yashvanth\OPCUA\opcuapython\opc_client\cert.pem,F:\Yashvanth\OPCUA\opcuapython\opc_client\key.pem")
    client.application_uri = "urn:example.org:FreeOpcUa:python-opcua"
    client.secure_channel_timeout = 10000
    client.session_timeout = 10000
    client.connect()
    root = client.get_root_node()
    objects = client.get_objects_node()
    print("childs og objects are: ", objects.get_children())`

//Error message

WARNING:opcua.crypto.security_policies:DEPRECATED! Do not use SecurityPolicyBasic128Rsa15 anymore!
WARNING:opcua.uaprotocol:Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security."))
CRITICAL:opcua.client.ua_client.Socket:Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security."))
ERROR:opcua.client.ua_client.Socket:Protocol Error
Traceback (most recent call last):
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 101, in _run
    self._receive()
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 121, in _receive
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "C:\Users\Wimera .LAPTOP-0OJQ2QSC\AppData\Roaming\Python\Python37\site-packages\opcua\client\ua_client.py", line 131, in _call_callback
    .format(request_id, self._callbackmap.keys())
opcua.ua.uaerrors._base.UaError: No future object found for request: 0, callbacks in list are dict_keys([1])
Traceback (most recent call last):
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\ptvsd_launcher.py", line 119, in <module>
    vspd.debug(filename, port_num, debug_id, debug_options, run_as)
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\debugger.py", line 37, in debug
    run(address, filename, *args, **kwargs)
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_local.py", line 64, in run_file
    run(argv, addr, **kwargs)
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_local.py", line 125, in _run
    _pydevd.main()
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_vendored\pydevd\pydevd.py", line 1752, in main
    globals = debugger.run(setup['file'], None, None, is_module)
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_vendored\pydevd\pydevd.py", line 1099, in run
    return self._exec(is_module, entry_point_fn, module_name, file, globals, locals)
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_vendored\pydevd\pydevd.py", line 1106, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\microsoft\python\core\Packages\ptvsd\_vendored\pydevd\_pydev_imps\_pydev_execfile.py", line 25, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)

AndreasHeine, 2020-06-01:
have you looked up the examples?
have you looked up prosys manual?

https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client_to_prosys_cert.py

AndreasHeine, 2020-06-01:
your problem is prosys!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

TESTSETUP: 

Prosys OPC UA Simulation Server 4.0.2-108 (On my Laptop "192.168.2.108")
[image]
[image]
your python client cert must be in the trusted/certs folder according to the prosys manual wich u dont read properly...
[image]

Python OPC UA Client (On my Surface)

```
from opcua import Client

client = Client("opc.tcp://192.168.2.108:4840/OPCUA/SimulationServer")
client.set_security_string("Basic256Sha256,SignAndEncrypt,cert.pem,key.pem")
client.application_uri = "urn:example.org:FreeOpcUa:python-opcua"
client.connect()
root = client.get_root_node()
objects = client.get_objects_node()
print("childs og objects are: ", objects.get_children())

'''
output:
...Python/OPC UA/prosys/prosys.py"
childs og objects are:  [Node(FourByteNodeId(i=2253)), Node(StringNodeId(ns=3;s=85/0:Simulation)), Node(StringNodeId(ns=5;s=MyObjectsFolder)), Node(StringNodeId(ns=6;s=StaticData))]
'''
```

do not run prosys server and python client on the same maschine...
if you change setting u must restart prosys server...
prosys crap...

yashvanth95, 2020-06-02:
> have you looked up the examples?
> have you looked up prosys manual?
> 
> https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client_to_prosys_cert.py
> 
> https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client_to_prosys_crypto.py

i have gone through these examples i read the prosys manual and in prosys simulation i trusted the python certs
i am using above examples only for connection with Prosys server 
if changes some credentials i will restart the prosys 
now i am using Basic256Sha256 
but still, i a, getting an error  
Like Bad_SecurityChecksFailed 
i have done all things you mentioned

AndreasHeine, 2020-06-02:
as shown above i was able to connect to prosys and read some nodes... 

you dont listen and you dont understand... i told you already how to generate a proper cert for your client: #1044

its an old repo with a regular x509 cert from the examples... not all stacks accept them this is something you have to look up -> prosys... i told you in #1044 not to use example certs!

> Bad_SecurityChecksFailed and BadCertificateUriInvalid
tells you all you need to know... your using a cert which has no subjAltname extention in it (x509**v3**) which you just copied from an example... or the URI and/or IP in the cert are different to your clients URI and/or IP!

https://forum.prosysopc.com/
ask them what kind of cert the accept and how to do it... i already did!

**i will not reply to this topic anymore**

yashvanth95, 2020-06-02:
sorry for causing inconvenient, i developed C++ opc ua client which was working fine  but suddenly i shifted to python here we need to create own certs before implementing but in C++ API  will take care that

closed 2020-06-02 by yashvanth95