FreeOpcUa/python-opcua #1052: BadSessionIdInvalid

mstfldmr, 2020-06-01:
Hello,

I am subscribing to 500 data points on Kepware server. It runs well but from time to time (may be once in a few days) I get BadSessionIdInvalid errors.

```
connection.py:332,Waiting for header
ua_client.py:334,read
connection.py:334,received header: Header(type:b'MSG', chunk_type:b'F', body_size:84, channel:3782932316)
ua_client.py:53,Sending: ReadRequest(TypeId:FourByteNodeId(i=631), RequestHeader:RequestHeader(AuthenticationToken:NumericNodeId(i=964719756), Timestamp:2020-05-31 00:49:21.146378, RequestHandle:154276, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:1000, AdditionalHeader:ExtensionObject(TypeId:TwoByteNodeId(i=0), Encoding:0, None bytes)), Parameters:ReadParameters(MaxAge:0, TimestampsToReturn:TimestampsToReturn.Source, NodesToRead:[ReadValueId(NodeId:FourByteNodeId(i=2259), AttributeId:AttributeIds.Value, IndexRange:None, DataEncoding:QualifiedName(0:None))]))
ua_client.py:475,call_publish_callback
ua_client.py:91,ServiceFault from server received while waiting for publish response
_base.py:326,exception calling callback for <Future at 0x7fe0efb2a590 state=finished returned Buffer>
Traceback (most recent call last):
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 483, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
connection.py:332,Waiting for header
connection.py:334,received header: Header(type:b'MSG', chunk_type:b'F', body_size:84, channel:3782932316)
ua_client.py:475,call_publish_callback
ua_client.py:91,ServiceFault from server received while waiting for publish response
opcua.ua.uaerrors._auto.BadSessionIdInvalid: "The session id is not valid."(BadSessionIdInvalid)_base.py:326,exception calling callback for <Future at 0x7fe0ef9f0290 state=finished returned Buffer>
Traceback (most recent call last):
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 483, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
connection.py:332,Waiting for header
connection.py:334,received header: Header(type:b'MSG', chunk_type:b'F', body_size:84, channel:3782932316)
connection.py:332,Waiting for header
ua_client.py:91,ServiceFault from server received  in response to ReadRequest
opcua.ua.uaerrors._auto.BadSessionIdInvalid: "The session id is not valid."(BadSessionIdInvalid)Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.7/dist-packages/opcua/client/client.py", line 70, in run
    val = server_state.get_value()
  File "/usr/local/lib/python3.7/dist-packages/opcua/common/node.py", line 153, in get_value
    result = self.get_data_value()
  File "/usr/local/lib/python3.7/dist-packages/opcua/common/node.py", line 162, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "/usr/local/lib/python3.7/dist-packages/opcua/common/node.py", line 273, in get_attribute
    result = self.server.read(params)
  File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 337, in read
    data = self._uasocket.send_request(request)
  File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 84, in send_request
    self.check_answer(data, " in response to " + request.__class__.__name__)
  File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionIdInvalid: "The session id is not valid."(BadSessionIdInvalid)
ua_client.py:103,Socket has closed connection
ua_client.py:109,Thread ended
```

How can I investigate the root cause of the problem? Is it possible to recover from here without losing data?

AndreasHeine, 2020-06-01:
do you use cisco vlan switches?

mstfldmr, 2020-06-01:
yes, I think so.

AndreasHeine, 2020-06-01:
ive got simmilar problems on a customers network... maybe check the logs from those switches and also the settings maybe something hints to a problem, make a wireshark trace aswell if there are a lot of arp requests in it you might have the same issues as i!

AndreasHeine, 2020-06-01:
have a look on docker instead of vm

mstfldmr, 2020-06-01:
It's a running production setup. All I have is a VM to collect data.

The error does not occur very often, so catching it with Wireshark is hard. Where can I get any other clues?

AndreasHeine, 2020-06-01:
have a look on docker instead of vm if possible!

the cisco switches can be configured in many ways and the have a errorlog.

AndreasHeine, 2020-06-01:
sometimes vm kind of stuck in processing like a lag ... u can test with a pc or laptop just subscribe same datapoint run and log the time when client connects! later you can filter opcua protocol and then in each response should be a timstamp so you can track down to your connection loss problem

mstfldmr, 2020-06-04:
since I can't recreate this error when I want, it's not easy to test.

as a workaround, I want to reconnect after getting this exception. where should I try/except for this exception in the client code?

AndreasHeine, 2020-06-04:
if you plan to auto reconnect:
https://github.com/AndreasHeine/opcua-sub-to-websocket/blob/master/opcua_client.py

mstfldmr, 2020-06-08:
Thanks @AndreasHeine