FreeOpcUa/python-opcua #1053: Call method with multiple input

andreafresa, 2020-06-01:
Hello,
I have a method with multiple input on my server.
The code is this one:
` inarg1 = ua.Argument()
    inarg1.Name = "Joint1"
    inarg1.DataType = ua.NodeId(ua.ObjectIds.Float)
    inarg1.ArrayDimension = []
    inarg2 = ua.Argument()
    inarg2.Name = "Joint2"
    inarg2.DataType = ua.NodeId(ua.ObjectIds.Float)
    inarg2.ArrayDimension = []
`
The Client is this one:
`
obj = client.get_node(id_obj)
 res = obj.call_method(node.nodeid.to_string(), [30.0 , 10.0])
`

Evrytime I try to run I get this error:
 File "C:\Users\edafenr\Desktop\python-opcua\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNoMatch: "The requested operation has no match to return."(BadNoMatch)

Can someone please help me? Thanks

oroulet, 2020-06-01:
One of your node id or path is wrong and does not exist in server

andreafresa, 2020-06-01:
I don´t think so because once I rerun my client the method run on my server (The function has a print inside and it shows on my pc).
I guess there is a problem in variables. Are there same examples somewhere on how to call a method with multiple input?

oroulet, 2020-06-01:
The server says so. And yes many. Read the examples before asking

andreafresa, 2020-06-01:
Solved. No node issue

closed 2020-06-01 by andreafresa