FreeOpcUa/python-opcua #1055: Application URI

yashvanth95, 2020-06-02:
1. self.application_uri = "urn:freeopcua:client"
2. self.product_uri = "urn:freeopcua.github.io:client"

the application URI and product URI as hardcoded in client.py, can we change according to Server URI? because it will be used when certs creation. 
  
3. self.security_policy = ua.SecurityPolicy() 
in SecurityPolicy class URI = "http://opcfoundation.org/UA/SecurityPolicy#None" can we add other security policies like  basic256 , basic256sha256. 

4. self.secure_channel_timeout = 3600000  # 1 hour
self.session_timeout = 3600000  # 1 hour
will the client will stop after 1hour timeout i don't want the time limit how to change the timeout 

5. I want to connect to the server with 
Basic256Sha256+SignAndEncrypt+username+password+certs happing 
Basic256Sha256+SignAndEncrypt+**no certs** but i searched the which API used but I didn't get can anyone help with these

AndreasHeine, 2020-06-02:
https://docs.python.org/3/tutorial/classes.html
-> 9.3.5. Class and Instance Variables

swamper123, 2020-06-02:
> 
>    self.application_uri = "urn:freeopcua:client"
>    self.product_uri = "urn:freeopcua.github.io:client"
>
>the application URI and product URI as hardcoded in client.py, can we change according to Server URI? because it will be used when certs creation.

What is stopping you to overwrite this attribute? Have a look at the Python Doc how to deal with such a thing.

> can we add other security policies like basic256 , basic256sha256.
No to basic256, because it is deprecated (but people complained that it was removed). And where do you want to add these anyway? Why not just use it?

>will the client will stop after 1hour timeout i don't want the time limit how to change the >timeout
Do you know what KeepAlive means & does? This timeout time seems fine.

> 
> Basic256Sha256+SignAndEncrypt+username+password+certs happing
> Basic256Sha256+SignAndEncrypt+no certs but i searched the which API used but I didn't get can anyone help with these
Sorry, but I don't understand what you mean.

yashvanth95, 2020-06-02:
keep alive used to check the connection is valid between 2 connected devices 
because this time-out was there in C++ SDK example  after 1hour client  stopped even the server was running 

i want to connect to the server with security mode and policy, username and  password without certs files 

 
> What is stopping you to overwrite this attribute? Have a look at the Python Doc how to deal with

if i want to connect to 10-15 servers i need to change the URI and again i have build

swamper123, 2020-06-02:
> keep alive used to check the connection is valid between 2 connected devices
> because this time-out was there in C++ SDK example after 1hour client stopped even the server was running

What will happen if your server hasn't got a timeout and your client hasn't got one? What will happen if you won't receive an anwer of the server? And what Client request are you making, that the server needs longer than an hour?

> if i want to connect to 10-15 servers i need to change the URI and again i have build

So you have to make a list of all uris and overwrite this attribute with each one, when you want to make a connection to a server.

AndreasHeine, 2020-06-02:
OPC UA security policys (message security):
```
    "None" 
    "Basic128Rsa15_Sign" Cert+Key necessary => DEPRECATED, do not use anymore!
    "Basic128Rsa15_SignAndEncrypt" Cert+Key necessary => DEPRECATED, do not use anymore!
    "Basic256_Sign" Cert+Key necessary
    "Basic256_SignAndEncrypt" Cert+Key necessary
    "Basic256Sha256_Sign" Cert+Key necessary
    "Basic256Sha256_SignAndEncrypt" Cert+Key necessary
```

OPC UA User Identity Token Types:
```
    "Anonymous" No identety check!
    "Basic256Sha256" identety via cert
    "Username" username and password
```

choose!

yashvanth95, 2020-06-02:
> OPC UA security policys (message security):
>
> ```
>     "None"
>     "Basic128Rsa15_Sign" Cert+Key necessary => DEPRECATED, do not use anymore!
>     "Basic128Rsa15_SignAndEncrypt" Cert+Key necessary => [...]

thanks for the reply i understand the above configuration 

if the server has Basic256Sha256 and SignAndEncrypt +username &password  without certs is this type of configuration is possible   

by above configuration can we connect to the server without generating certs and updating URI in
self.application_uri = "urn:freeopcua:client"
self.product_uri = "urn:freeopcua.github.io:client"

AndreasHeine, 2020-06-02:
> if the server has Basic256Sha256 and SignAndEncrypt +username &password without certs is this type of configuration is possible

OPC UA security policys (message security):
```

    "None" 
    "Basic128Rsa15_Sign" Cert+Key necessary => DEPRECATED, do not use anymore!
    "Basic128Rsa15_SignAndEncrypt" Cert+Key necessary => DEPRECATED, do not use anymore!
    "Basic256_Sign" Cert+Key necessary
    "Basic256_SignAndEncrypt" Cert+Key necessary
    "Basic256Sha256_Sign" Cert+Key necessary
    "Basic256Sha256_SignAndEncrypt" Cert+Key necessary
```

OPC UA User Identity Token Types:
```

    "Anonymous" No identety check!
    "Basic256Sha256" identety via cert
    "Username" username and password
```

choose again (-_-)

yashvanth95, 2020-06-02:
OPC UA User Identity Token Types:
Basic256sha256 + username and password 
why i am choosing because the server has this type of configuration that why

AndreasHeine, 2020-06-02:
you just need to choose on message security mode (security policys) and one for identety check (User Identity Token Types)

yashvanth95, 2020-06-02:
> you just need to choose on message security mode (security policys) and one for identety check (User Identity Token Types)

if we choose basic256sha256 is certs are necessary with user name and password then in server side we need to make trust  that certs right

AndreasHeine, 2020-06-02:
the only no cert options are:

```
security policys : "None" 
User Identity Token Types: "Anonymous"
```

```
security policys : "None" 
User Identity Token Types: "Username"
```

any other configuration needs cert and key for encryption or auth purposes

yashvanth95, 2020-06-02:
thanks, i got clarity  now
when we choose this configuration then we make it as a trust on server-side right 
subjectAltName = URI:urn:opcua:python:server,IP: 127.0.0.1
in IP can give the server IP or localhost on which the client is running

AndreasHeine, 2020-06-02:
URI and IP should match your client application aswell as the subjectAltname in the Cert so the server can validate your client + prosys only accepts trustet certs!

yashvanth95, 2020-06-02:
> URI and IP should match your client application aswell as the subjectAltname in the Cert so the server can validate your client + prosys only accepts trustet certs!

i have given IP address of my pc where my client and prosys are running in same PC 
if the prosys in another PC and client in another pc 
then the IP address will be where the client running right

AndreasHeine, 2020-06-02:
client cert => clients uri and ip
server cert => servers uri and ip (if its prosys they took care on it)

yashvanth95, 2020-06-03:
do i need to create own application_uri in client and that will be add in the generation of certs 
or de we need application_uri of server

AndreasHeine, 2020-06-03:
if your application consists of only one client class instance you need only on client uri, never use server uri in client cert or even server uri as client uri... its like copying someones passport and say hi i am also this application... doeasn't make sence! Each Client or Server instance needs a unique URI!

yashvanth95, 2020-06-03:
> What will happen if your server hasn't got a timeout and your client hasn't got one? What will happen if you won't receive an anwer of the server? And what Client request are you making, that the [...]

do i need to make changes in self.application_uri = "urn:freeopcua:client"  if need to connect to 10-15 servers
 i am using name URI for generation for certs 
In siemens machine which has opc server, how to trust the client certs on that server-side do you have any idea because i have gone through this link [https://github.com/AndreasHeine/PythonOpcUaServer-for-Simatic-S7-1500-OpcUaClient]

AndreasHeine, 2020-06-03:
> do i need to make changes in self.application_uri = "urn:freeopcua:client" if need to connect to 10-15 servers
set it whatever you put in your cert as uri before connectiong

S7-1500? WinCC? is Siemens Client or Server? please be more precise!!!

yashvanth95, 2020-06-03:
> > do i need to make changes in self.application_uri = "urn:freeopcua:client" if need to connect to 10-15 servers
> > set it whatever you put in your cert as uri before connectiong
> 
> S7-1500? WinCC? is Siemens Client or Server? please be more precise!!!

i am implementing client and server is siemens Simatic I don't which series

AndreasHeine, 2020-06-03:
i highly reccomend you reading the siemens opc ua docs for the products you use!

i guess this topic "Application URI" can be closed!

yashvanth95, 2020-06-03:
yes last question 

> > > do i need to make changes in self.application_uri = "urn:freeopcua:client" if need to connect to 10-15 servers in client side 

> > > set it whatever you put in your cert as uri before connectiong

>>>So you have to make a list of all uris and overwrite this attribute with each one, when you want to make a connection to a server.

i got confused from the above reply 
if  i need to connect to the client to 10-15 servers i don't need change the Uri Right the URL will change

AndreasHeine, 2020-06-03:
if you plan to connect to 10 different servers at the same time you schould use 10 Client instances, each of this instances can have the same URI and CERT because its physically one application in one python process!

yashvanth95, 2020-06-03:
@AndreasHeine  thank so much you for clearing the doubts whether it is silly also.

closed 2020-06-03 by yashvanth95