FreeOpcUa/python-opcua #1057: .../opcua/common/node.py => Node class => __str__ and __repr__ method

AndreasHeine, 2020-06-02:
Hey guys i plan to make a PR on Nodeclass as seen from the topic its about the __str__ and the __repr__ methods

https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/common/node.py
```
    def __str__(self):
        return "Node({0})".format(self.nodeid)
    __repr__ = __str__
```

what i like to change is that if you cast a node to string 
Now: "str(node)" & "repr(node)" => Node(NumericNodeId(ns=2;i=2))
After: "str(node)" => "ns=2;i=2" and if you call repr(node) => Node(NumericNodeId(ns=2;i=2))

what is your opinion?

@oroulet

oroulet, 2020-06-02:
Sounds correct. Having str and repr the same is a bit of an abuse

AndreasHeine, 2020-06-02:
@oroulet => #1058 tell me if its not ok

AndreasHeine, 2020-06-03:
`__repr__ = __str__`

is used in many places in the same way i could change them in the same manner if you like!?

swamper123, 2020-06-03:
> 
> 
> `__repr__ = __str__`
> 
> is used in many places in the same way i could change them in the same manner if you like!?

This seems to be usefull, especially when you rewrite both.

oroulet, 2020-06-03:
This makes sense for nodeid since they have a specified string representation. But do other objects have such a representation?

AndreasHeine, 2020-06-03:
maybe Browsname would make sence

of cource only in these cases where it makes sence

oroulet, 2020-06-03:
Ok.

closed 2020-06-03 by AndreasHeine