FreeOpcUa/python-opcua #1059: Prosys Client Security Mode "None" can't browse nodes

su600, 2020-06-02:
Hi, I am facing error INFO as follow:
```
WARNING:opcua.binary_server_asyncio:processor returned False, we close connection from ('127.0.0.1', 56554)
INFO:opcua.binary_server_asyncio:Lost connection from ('127.0.0.1', 56554), None
Cleanup client connection:  ('127.0.0.1', 56554)
```
This happens when I use Prosys OPC UA Client try to browse the information model by NoSecurity mode(server-minimal.py) . By using Sign & SignAndEncrypt mode is fine, and UaExpert is fine without problems.

Could you please help me figure out what does this mean and what should I do to fix this?

Thank you!

AndreasHeine, 2020-06-02:
i tested it with my laptop and prosys android client and i was able to browse the server-minimal.py
what client do you use?
what i realised with prosys that server and client on the same maschine makes trouble some times but you could try to use the real ip of your Ethernet or WiFi instead of localhost or 0.0.0.0

AndreasHeine, 2020-06-04:
@su600 solved?

su600, 2020-06-05:
@AndreasHeine Hi!  Thanks your for replay, I tried your tips, but I still can't browse the information model by **Prosys Client through None security mode**.

I think my problem maybe a problem of **security setting or certificate**

When I load certificate and private_key like this:

```
await server.load_certificate("certificate.der")
await server.load_private_key("private-key.pem")

server.set_security_policy([
            ua.SecurityPolicyType.NoSecurity,
            ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt,
            ua.SecurityPolicyType.Basic256Sha256_Sign])
```
Prosys client **can** browse information model well through SignAndEncrypt, **but None Security can't.**
The server returns:
```
WARNING:opcua.binary_server_asyncio:processor returned False, we close connection from ('127.0.0.1', 56554)
INFO:opcua.binary_server_asyncio:Lost connection from ('127.0.0.1', 56554), None
Cleanup client connection:  ('127.0.0.1', 56554)
```
It is strange that UaExpert is always can browse information model both `None `and `SignAndEncrypt` 

Maybe I am new to OPC UA standards or Python-OPC UA (or is a Bug), I still try to figure out how to make Prosys Client browse information model through Node Security mode.

Do you have any idea to fix this? 

Thank you again !

AndreasHeine, 2020-06-05:
UaExpert is a forgiving peace of software!

The strange thing i am able to browse in "None", what i had experienced is that sometimes Prosys-Products behave different on other pc's so i am guessing that there might be some settings in the os different maybe firewall or other stuff.

I testet Prosys Browser v4.0.4-247 SDK Version 4.3.0-1075 with an empty python server instance:

```
from opcua import Server
server = Server()
server.set_endpoint("opc.tcp://127.0.0.1:4840")
server.start()
```

python-opcua version: opcua-0.98.9 and opcua-0.98.11

oroulet, 2020-06-05:
jus tested server-minimal.py and aserver-with.encyrptions with prosys client and it works fine on my PC

TobiasNils, 2022-04-19:
Hi @oroulet, how/which peer certificates do you apply on the server side to make this work?