FreeOpcUa/python-opcua #1062: BadTcpMessageTooLarge   description="Chunk size (132) exceeded maximum

yashvanth95, 2020-06-04:
i am getting  this error when connecting to prosys server with Security and policy mode or with NONE
i have gone though this link [https://forum.prosysopc.com/forum/opc-ua-simulation-server/bad_tcpmessagetoolarge-code0x80800000-descriptionchunk-size-132-exceeded-maximum-1/]() they saying the problem is with python  opc ua

Received an error: MessageAbort(error:StatusCode(BadTcpMessageTooLarge), reason:Bad_TcpMessageTooLarge (code=0x80800000, description="Chunk size (132) exceeded maximum (-1)"))
Protocol Error

anyone know what is the reason for this error

AndreasHeine, 2020-06-04:
and i told u in #1051

> do not run prosys server and python client on the same maschine...

yashvanth95, 2020-06-04:
it was connecting yesterday i run the client and server whole day i  run today it is giving error nothing has changed in client or server side

AndreasHeine, 2020-06-04:
> it was connecting yesterday i run the client and server whole day i run today it is giving error nothing has changed in client or server side

then its more likely a problem with your computer!

AndreasHeine, 2020-06-04:
> if i am running in the server in 1 PC and client in 1 PC , what will be the URL
> for example server running PC will have IP 192.168.0.1 and client PC IP 192.168.0.2 in same network
> then to connect to there server should i give the IP 192.168.0.1?

`protokoll://domainorip:port/path`

please fokus a little more on the OPC UA basics

yashvanth95, 2020-06-04:
but when i was getting the data from the server for a whole day  because of that's the reason  thanks for the solution

>>protokoll://domainorip:port/path
>>please fokus a little more on the OPC UA basics
can you give the example for this

AndreasHeine, 2020-06-04:
https://github.com/FreeOpcUa/python-opcua/blob/f761ccf0882daff8f000885e851db66b3c0ea167/examples/client-example.py

yashvanth95, 2020-06-04:
but when run the opc ua c++ server i am getting data and connection is shaping but when i run the prosys server i am getting errors

AndreasHeine, 2020-06-04:
@oroulet : 0x80800000 => The size of the message specified in the header is too large
it occures when you use "localhost" prosys-server and python client
could you have a look -> [uaprotocol_hand.py](https://github.com/FreeOpcUa/python-opcua/blob/f761ccf0882daff8f000885e851db66b3c0ea167/opcua/ua/uaprotocol_hand.py) -> no clue looks ok!

@yashvanth95 : might be something with the java sdk ... i tested it against many industrial products and python opcua woks just fine!

yashvanth95, 2020-06-04:
i tried running server and client in different  machines but the server is connecting to in uaexpert but still getting the same error i tried with None in uaexpert

//sample code
   `from opcua import Client

     url = "opc.tcp://192.168.1.132:4840/OPCUA/SimulationServer"
    client = Client(URL)
    client.connect()
    servicelevel_node = client.get_node("ns=0;i=2267")
    servicelevel_node_value = servicelevel_node.get_value()
    print(servicelevel_node_value)
    EndnodepointValue = client.get_node("ns=0;i=2255")    
     print("EndnodepointValue",EndnodepointValue.get_value() )
    
        `

AndreasHeine, 2020-06-04:
sorry i am unable reproduce the behavior!

yashvanth95, 2020-06-04:
> sorry i am unable reproduce the behavior!

i didn't get 
is there any other solution for this

yashvanth95, 2020-06-04:
i got the response when i changed the values in 
 opcua.ua.uaprotocol_hand
        self.ProtocolVersion = 0
        self.ReceiveBufferSize = 65536
        self.SendBufferSize = 65536
        self.MaxMessageSize = 16777216
        self.MaxChunkCount = 5000 
        self.EndpointUrl = ""
        self._freeze = True

according to this link will this affect the connection and creation of session and subscription   [https://stackoverflow.com/questions/45687549/freeopcua-python-client-example](url)

AndreasHeine, 2020-06-04:
https://reference.opcfoundation.org/v104/Core/docs/Part6/7.1.2/

yashvanth95, 2020-06-04:
how to handle this type of large messages?

AndreasHeine, 2020-06-04:
i dont know python client sends it capabilities inside the "hello" from then its inside prosys and i dont know anything about what stack and so on...

yashvanth95, 2020-06-04:
ok thanks again for clearing doubts

closed 2020-06-05 by yashvanth95