FreeOpcUa/python-opcua #1064: Accessing Historical Data Access ( HDA ) from SIEMENS SCADA using Free OPCUA .

rajnishnationfirst, 2020-06-05:
Hi Everyone ,

**I want to read the HISTORICAL DATA ( HDA ) from WinCC Siemens SCADA.**

I came to know that we have to use the below code :

        **myparams = ua.HistoryReadParams()
        #myparams.x = ???
        #myparams.someotherattr = ???
        myhistory = client.history_read(myparams)**

**Final Code :**

```
import sys
sys.path.insert(0, "..")

import logging
import time

from ast import literal_eval as make_tuple

try:
    from IPython import embed
except ImportError:
    import code

    def embed():
        vars = globals()
        vars.update(locals())
        shell = code.InteractiveConsole(vars)
        shell.interact()

from opcua import Client
from opcua import ua
from opcua import uaprotocol_auto

import requests

class Siemens_HDA_Client(object):

 
    def __init__(self):
        print('Before Connection')
        client = Client("opc.tcp://EC2AMAZ-VJQV23P:4862")
        print('After Execution')

    def get_HDA_Data_From_Siemens_Scada(self):
        myparams = ua.HistoryReadParams()
        #myparams.x = ???
        #myparams.someotherattr = ???
        myhistory = client.history_read(myparams)

p1 = Siemens_HDA_Client()

print('Data')    

print(p1.get_HDA_Data_From_Siemens_Scada())
```

[image]

If anyone can help me out on this .
I am not able to locate the module.

Thanks and Regards
Rajnish Vishwakarma

AndreasHeine, 2020-06-05:
https://github.com/FreeOpcUa/python-opcua/blob/f761ccf0882daff8f000885e851db66b3c0ea167/opcua/ua/uaprotocol_auto.py#L8322

rajnishnationfirst, 2020-06-05:
Hi Andreas ,

I came to know that history_read is a function which are in UA_client and Node modules , i am using the clientof histtory_read() , i m getting the error mentioned below :

I changed my code to :

```
import sys
sys.path.insert(0, "..")

import logging
import time

from ast import literal_eval as make_tuple

try:
    from IPython import embed
except ImportError:
    import code

    def embed():
        vars = globals()
        vars.update(locals())
        shell = code.InteractiveConsole(vars)
        shell.interact()

from opcua import Client
from opcua import ua

import requests

class Siemens_HDA_Client(object):

 
    def __init__(self):
        print('Constructor Called')
        

    def get_WinCC_Siemens_HDA_Client_Connection(self):
        client = Client("opc.tcp://EC2AMAZ-VJQV23P:4862")
        return client
        

            
    def get_HDA_Data_From_Siemens_Scada(self):
        myparams = ua.HistoryReadParameters()
        print(myparams)
        #myparams.x = ???
        #myparams.someotherattr = ???

        hda_client=self.get_WinCC_Siemens_HDA_Client_Connection()
        try:
            hda_client.connect()
            node=hda_client.get_node("ns=2;i=53")
            print('Rajnish')
        finally:
            print("Finally Block")
        
        myhistory = hda_client.history_read(myparams)

p1 = Siemens_HDA_Client()

print('Data')

p1.get_HDA_Data_From_Siemens_Scada()

```
Now i am getting the below error ::

AttributeError: 'Client' object has no attribute 'history_read'

[image]

brubbel, 2020-06-19:
https://github.com/FreeOpcUa/python-opcua/blob/34d98f992f9f7e494be20f028cdf9a05b3b62540/opcua/common/node.py#L492-L498

Use read_raw_history from the node object. This will construct the history parameters for you:

```
node = hda_client.get_node(...)
dvList = node.read_raw_history(numvalues=10)
print([dv.Value.Value for dv in dvList])
```

duduyoyo, 2024-02-01:
Check [here](https://github.com/duduyoyo/WebSocket4OPC), it works with classic OPC HDA/DA/AE. Cheers!