FreeOpcUa/python-opcua #1065: re-establish connection with server and creating new session

ganesh1306, 2020-06-05:
i have gone through this link [https://reference.opcfoundation.org/v104/Core/docs/Part4/6.7/](url)
but didn't get much when the server gets disconnected i need to reconnect and subscribe 
when the connection successes their is no status of connection or failure  without losing information

       client = Client("opc.tcp://localhost:4840/freeopcua/server/")
       client.connect()
can any one have solution for this

AndreasHeine, 2020-06-05:
if the server stopps/breaks/restart the client has to reastablish the connection most vendors read cyclic the servcelevel-node (ns=0;i=2267) if it fails to read your client knows the server is off and your client can attempt a reconnect and re-subscibe the nodes of interest. if you dont know how to do, you can check out this [client example](https://github.com/AndreasHeine/opcua-sub-to-websocket/blob/master/opcua_client.py)!

Hope it was helpfull.

ganesh1306, 2020-06-05:
it so helpful
don't be mad because i am new to python and OPC UA
what are asyncio means is like multi-threading

what is the below line code will do 

            start_server = websockets.serve(ws_handler=ws_handler, host="127.0.0.1", port=8000)
            asyncio.ensure_future(start_server, loop=loop)
            asyncio.ensure_future(main(client=client, subscription=subscription, 
            nodes_to_subscribe=nodes_to_subscribe, events_to_subscribe=events_to_subscribe,                                      handler=handler), loop=loop)

why ws_handler() function used 
service_level >= 200:
            
          subscribe(client=client, subscription=subscription, nodes_to_subscribe=nodes_to_subscribe, 
            events_to_subscribe=events_to_subscribe, handler=handler)
can this line will subscribe to DA and A&E 

where this "NotificationMessage is None". is printing from

AndreasHeine, 2020-06-05:
asyncio is a python module you should check the doc before using it! asyncio is one of three ways to do it 1. Multi-Threading 2. Multi-Processing 3. Async-Progaming -> But its a generall topic in python not related to OPC UA!

for connection monitoring/testing you can try read any node you want but i recommend using one which is in the Standard-Server-Object -> i always use service_level its part of the [opcua docs about redundancy servers](https://reference.opcfoundation.org/v104/Core/docs/Part4/6.6.2/)

>       subscribe(client=client, subscription=subscription, nodes_to_subscribe=nodes_to_subscribe, 
>         events_to_subscribe=events_to_subscribe, handler=handler)
> 
> can this line will subscribe to DA and A&E

if you want to subscribe nodes look in the examples: https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py

```
class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)

handler = SubHandler()
sub = client.create_subscription(500, handler)
handle = sub.subscribe_data_change(myvar)
```

ganesh1306, 2020-06-05:
>>def datachange_notification(self, node, val, data):
 >>   print("Python: New data change event", node, val)

>>    def event_notification(self, event):
>>        print("Python: New event", event)

when can we use this function

swamper123, 2020-06-05:
@ganesh1306 FYI:
>  what are asyncio means is like multi-threading

It is concourrency. There is a good [Video ](https://www.youtube.com/watch?v=GSiZkP7cI80)which explains how async stuff works with asyncio and the differences.

The docs will show you how to use stuff, but not directly how it works. But no worries, it isn't magic.

ganesh1306, 2020-06-05:
thanks for the quick reply @swamper123

AndreasHeine, 2020-06-05:
```
        def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

        def event_notification(self, event):
        print("Python: New event", event)

when can we use this function
```

its like a callback! whenever a datachange happens this function will be called!

ganesh1306, 2020-06-05:
>> its like a callback! whenever a datachange happens this function will be called!

the data changes will DA/A&E /HDA

AndreasHeine, 2020-06-05:
depends one what is subscribet to! 

the example has only datachange_notification and events (A&E) if you need more the 

> ws_handler () function use

websocket_handler => its not related to OPC UA it was a fun project to get subscriptiondata do a html page :)

> will this subscribe(client, subscription, nodes_to_subscribe, events_to_subscribe, handler): will handle all subscription to DA or A&E or HDA

no! just datachange_notification and event_notification read the docs and look at the examples...

ganesh1306, 2020-06-05:
will subscribe and unsubscribe will affect the server session count in real-time machines and demo serves

AndreasHeine, 2020-06-05:
> When a Client loses the connection to the server, the goal is to reconnect without losing information. To do this the Client shall re-establish the connection by creating a new SecureChannel and activating the Session with the Service ActivateSession

i guess its not implementet this way but i might be wrong!

you can also make it in a different way if you look at the whole monitored item methods in
https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/common/subscription.py

ganesh1306, 2020-06-05:
> i guess its not implementet this way but i might be wrong!
>
> you can also make it in a different way if you look at the whole monitored item methods in [...]

can we add in subscription.py 

        def datachange_notification(self, node, val, data):
         self.node = node
         self.val = val
         self.data = data
         print(node, val, data)
    how to use this publish_callback()

AndreasHeine, 2020-06-06:
> can we add in subscription.py
```
    """
    Subscription object returned by Server or Client objects.
    The object represent a subscription to an opc-ua server.
    This is a high level class, especially subscribe_data_change
    and subscribe_events methods. If more control is necessary look at
    code and/or use create_monitored_items method.
    """
```
you can make a PR if u like!
  
  

> how to use this publish_callback()

```
class SubHandler(object):
    """
    Subscription Handler. To receive events from server for a subscription
    This class is just a sample class. Whatever class having these methods can be used
    """

    def data_change(self, handle, node, val, attr):
        """
        Deprecated, use datachange_notification
        """
        pass

    def datachange_notification(self, node, val, data):
        """
        called for every datachange notification from server
        """
        pass

    def event_notification(self, event):
        """
        called for every event notification from server
        """
        pass

    def status_change_notification(self, status):
        """
        called for every status change notification from server
        """
        pass
```
http://documentation.unified-automation.com/uasdkhp/1.0.0/html/subscription_model.png

> called for every event notification from server

ganesh1306, 2020-06-08:
https://github.com/AndreasHeine/opcua-sub-to-websocket/blob/master/opcua_client.py

you are using asyncio instead can we use threading