FreeOpcUa/python-opcua #1066: No name 'ua' in module 'opcua'

xuplus6493, 2020-06-05:
Hello,
I have install opcua
but when I use this code by Visual Code
`
from opcua import Client,ua
`
I get the problem

No name 'ua' in module 'opcua' (no-name-in-module)

How can I fix it ?

AndreasHeine, 2020-06-05:
have you checked the folders? C:\Users\<your login>\AppData\Local\Programs\Python\Python37\Lib\site-packages\opcua
[image]

oroulet, 2020-06-05:
Your installation is completely broken. Try again

xuplus6493, 2020-06-05:
> have you checked the folders? C:\Users<your login>\AppData\Local\Programs\Python\Python38\Lib\site-packages\opcua
> [image]

yes，I have this file

AndreasHeine, 2020-06-05:
try:
python -m pip uninstall opcua
and finaly
python -m pip install opcua

xuplus6493, 2020-06-05:
> Your installation is completely broken. Try again

I have try 
pip uninstall opcua
pip install opcua

AndreasHeine, 2020-06-05:
(-_-) do you have python 2 also installed ?

xuplus6493, 2020-06-05:
> (-_-) do you have python 2 also installed ?

no, only python 3

AndreasHeine, 2020-06-05:
have you tried comandprompt with admin priviliges and then => python -m pip uninstall opcua =>  python -m pip install opcua

xuplus6493, 2020-06-05:
> have you tried comandprompt with admin priviliges and then => python -m pip uninstall opcua => python -m pip install opcua

I did, but it's not working

[image]

xuplus6493, 2020-06-05:
> Your installation is completely broken. Try again

I have reinstall python, visual code and opcua ...

AndreasHeine, 2020-06-05:
and? win/fail?

xuplus6493, 2020-06-05:
fail ...

AndreasHeine, 2020-06-05:
open python shell:
print( help('modules'))

xuplus6493, 2020-06-05:
> open python shell:
> print( help('modules'))

[image]
[image]

AndreasHeine, 2020-06-05:
if u use pythons builtin IDE -> "idle" does it also blame that there is no "ua"

xuplus6493, 2020-06-05:
> if u use pythons builtin IDE -> "idle" does it also blame that there is no "ua"
[image]

xuplus6493, 2020-06-05:
I want to writ Tag  (kepware server)
the same code is work on another computer
[image]

AndreasHeine, 2020-06-05:
yeah because ua.VariantType.Double is missing... then you'll get BadTypeMismatch !!!

xuplus6493, 2020-06-05:
> yeah because ua.VariantType.Double is missing... then you'll get BadTypeMismatch !!!

How to get  ua.VariantType.Double ?

my tag data type is "Double"

AndreasHeine, 2020-06-05:
you will not like my answer...

`from opcua import ua`

xuplus6493, 2020-06-05:
> you will not like my answer...
> 
> `from opcua import ua`

[image]

xuplus6493, 2020-06-05:
> opcua datatypes and python datatypes are different so you have to be specific what opcuatype you want to set!
> 
> "var = 1" is an Python int => OPCUA Int 64

But, the same code is working in another computer....

AndreasHeine, 2020-06-05:
well then its more likely your pc is the problem...

xuplus6493, 2020-06-05:
> well then its more likely your pc is the problem... sorry

Thank you for your help !

xuplus6493, 2020-06-07:
I solve this problem !!
pip install  copyright

closed 2020-06-07 by xuplus6493

gauravbvelex, 2024-07-21:
@xuplus6493  Could you please provide steps about how did you fix it? "pip install copyright" is vague.