FreeOpcUa/python-opcua #1067: how to use it in the arm-linux

KingDamLiu, 2020-06-08:
In my question,I can't cross compilling the lxml

zerox1212, 2020-06-08:
I don't think this is a question for here. You need to get lxml to work with your platform. Read the docs for lxml, it has other dependencies.

swamper123, 2020-06-08:
[[or use this quick link to their page](https://lxml.de/installation.html)]
libxml2 and libxslt has to be installed first.

KingDamLiu, 2020-06-08:
of course! I have installed libxml2 and libxslt.but I don't kown how to setting the parameter to the command "python3 setup.py build" inlcude arm-linux-gcc and the any path.

brubbel, 2020-06-20:
Can't you install via pip? It should build lxml for you if your specific arm architecture has no precompiled binaries available.