FreeOpcUa/python-opcua #1068: Reconnection with server after disconnection/shutdown

yashvanthb, 2020-06-09:
client  = Client(url)
client.connect() 
1. how to know using this API it is connected to the server and how to reconnect to the server.
i am using prosys server simulator https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py  i have gone through this link but the problem is from server-side i am getting The server has reached its maximum number of sessions.(BadTooManySessions) i know this from prosys server 
2. what and why this session count without incrementing the session count can we reconnect to the server 
3. will this session count will appear in realtime PLC servers

AndreasHeine, 2020-06-09:
> The server has reached its maximum number of sessions.(BadTooManySessions) 

server session timeout is to high... most have default of one hour... try 30s instead!
but this appears a prosys server simulator problem not python-opcua...

and the rest way to general or not related to python-opcua or this topic!

**please use stackoverflow or something else "issue" is for real issues with the code and specific questions**

yashvanthb, 2020-06-09:
>> i want to connect to the server when it is disconnect or shutdown only I don't want to monitor or connect  every time 
using NodeId ns=0;i=2259 when the disconnect happens then the client has to try the connection to a server for is an example of implantation

AndreasHeine, 2020-06-09:
subscribe ns=0;i=2259 and with the result you can du whatever you want!

if the server shutdown gracefully and if its statechange is implemented you will receive a notification about valuechange on server state then you can disconnect, but if the server is back up and running the client has to recconect (try client.connect() if it fails wait try again). 

> OPC UA Sessions are not persistent. I.e. when a Server restarts, all Sessions are lost. restoring / reconnecting in this situation is not possible.

https://documentation.unified-automation.com/uasdkhp/1.0.0/html/group__ua__base__serverstate.html
> ua_serverstate {    UA_SERVERSTATE_RUNNING = 0,  UA_SERVERSTATE_FAILED = 1,  UA_SERVERSTATE_NOCONFIGURATION = 2,  UA_SERVERSTATE_SUSPENDED = 3,     UA_SERVERSTATE_SHUTDOWN = 4,  UA_SERVERSTATE_TEST = 5,  UA_SERVERSTATE_COMMUNICATIONFAULT = 6,  UA_SERVERSTATE_UNKNOWN = 7   }

even UaExpert and other Clients like Siemens read cyclic in order to monitor connection-status!
there is no mechanism in opcua (just two examples how it could work) for that because its always depending on the server implementation so **you** have to adapt to the server (it always depend which parts of the specs are implemented serverside...)

yashvanthb, 2020-06-09:
i have implemented the connection logic but above errors when the client disconnects and shutdown is this the right method? if not the only way is a subscription?

https://github.com/FreeOpcUa/python-opcua/issues/1068#issuecomment-641100733

AndreasHeine, 2020-06-09:
```
while 1:
try:
client.connect()
Serverstatus = client.get_node("ns=0;i=2259").get_value()
print(Serverstatus)
except:
client.disconnect()
client.connect()
print(Serverstatus)
print(client.connect_socket())
```

do you know what youve done? (0_0)°

yashvanthb, 2020-06-09:
> ```
> while 1:
> client.connect()
> try:
> Serverstatus = client.get_node("ns=0;i=2259").get_value()
> print(Serverstatus)
> except:
> client.disconnect()
> client.connect()
> print(Serverstatus)
> print(client.connect_socket())
> ```
> 
> do you know what youve done? (0_0)°

i have modified now will this work
   
     client.connect()
            
    while 1:
     try:
          Serverstatus = client.get_node("ns=0;i=2259").get_value()
        print(Serverstatus)
    except:
        if(Serverstatus != 0):
            print("Clinet trying to connect to URL ",url)
            client.connect()
   
        
        sleep(2)

swamper123, 2020-06-09:
>  client.connect()
>         
> while 1:
>  try:
>       Serverstatus = client.get_node("ns=0;i=2259").get_value()
>     print(Serverstatus)
> except:
>     if(Serverstatus != 0):
>         print("Clinet trying to connect to URL ",url)
>         client.connect()
> 
>     
>     sleep(2)

This will work as long as you receive a Serverstatus value.
In the moment you disconnect, the value of Serverstatus will never change and stuck forever.
Add `Serverstatus=None` under your while statement and elif it in your exception.
Other way could be like @AndreasHeine mentioned, just indent them correctly (I'm not sure if a client is able to disconnect if `get_value` fails, but I trust you here Andreas ;) ).

AndreasHeine, 2020-06-09:
why is it so hard to just do something like "state machine":

case = 0
while 1:
if case == 0:
    init client
    case=1

if case ==1:
try:
    connect
    case = 2
except:
    wait some time
    case = 1

if case ==2:
    try:
        read node
        wait some time
    except:
        connection is broken 
... and so on....

yashvanthb, 2020-06-09:
> why is it so hard to just do something like "state machine":
> 
> ```
> 
> case = 0
> while 1:
> if case == 0:
>     init client
>     case=1
> 
> if case ==1:
> try:
>     connect
>     case = 2
> except:
>     wait some time
>     case = 1
> 
> if case ==2:
>     try:
>         read node
>         wait some time
>     except:
>         connection is broken 
> ```
> 
> ... and so on....
> 
> told you already... its async example but if you remove async and await it will work in sync the same way:
> https://github.com/AndreasHeine/opcua-sub-to-websocket/blob/master/async_client.py
> in "async def opcua_client():"
> 
> ```
> async def opcua_client():
>     """
>     -handles connect/disconnect/reconnect/subscribe/unsubscribe
>     -connection-monitoring with cyclic read of the service-level
>     """
>     client = Client(url=server_url)
>     handler = SubscriptionHandler()
>     subscription = None
>     case = 0
>     subscription_handle_list = []
>     idx = 0
>     while 1:
>         if case == 1:
>             #connect
>             print("connecting...")
>             try:
>                 await client.connect()
>                 await client.load_type_definitions()
>                 idx = await client.get_namespace_index("http://andreas-heine.net/UA")
>                 print("connected!")
>                 case = 2
>             except:
>                 print("connection error!")
>                 case = 1
>                 await asyncio.sleep(5)
>         elif case == 2:
>             #subscribe all nodes and events
>             print("subscribing nodes and events...")
>             try:
>                 variable_list = await client.get_node("ns=2;i=6").get_children() # added for performance test with 100 quick and randomly changing variables
>                 subscription = await client.create_subscription(50, handler)
>                 subscription_handle_list = []
>                 if nodes_to_subscribe:
>                     for node in nodes_to_subscribe + variable_list: # added for performance test with 100 quick and randomly changing variables
>                     # for node in nodes_to_subscribe:
>                         handle = await subscription.subscribe_data_change(client.get_node(node))
>                         subscription_handle_list.append(handle)
>                 if events_to_subscribe:
>                     for event in events_to_subscribe:
>                         handle = await subscription.subscribe_events(event[0], event[1])
>                         subscription_handle_list.append(handle)
>                 print("subscribed!")
>                 case = 3
>             except:
>                 print("subscription error")
>                 case = 4
>                 await asyncio.sleep(0)
>         elif case == 3:
>             #running => read cyclic the service level if it fails disconnect and unsubscribe => wait 5s => connect
>             try:
>                 if users == set():
>                     datachange_notification_queue.clear()
>                     event_notification_queue.clear()
>                 service_level = await client.get_node("ns=0;i=2267").get_value()
>                 if service_level >= 200:
>                     case = 3
>                 else:
>                     case = 4
>                 await asyncio.sleep(5)
>             except:
>                 case = 4
>         elif case == 4:
>             #disconnect clean = unsubscribe, delete subscription then disconnect
>             print("unsubscribing...")
>             try:
>                 if subscription_handle_list:
>                     for handle in subscription_handle_list:
>                         await subscription.unsubscribe(handle)
>                 await subscription.delete()
>                 print("unsubscribed!")
>             except:
>                 print("unsubscribing error!")
>                 subscription = None
>                 subscription_handle_list = []
>                 await asyncio.sleep(0)
>             print("disconnecting...")
>             try:
>                 await client.disconnect()
>             except:
>                 print("disconnection error!")
>             case = 0
>         else:
>             #wait
>             case = 1
>             await asyncio.sleep(5)
> ```

i tried the above examples and logic but the maximum session is exceeded error from the server

AndreasHeine, 2020-06-09:
i used the logic in production and no issues at all... then you made something wrong...

yashvanthb, 2020-06-09:
i am using prosys server

AndreasHeine, 2020-06-09:
cant help with prosys server, dont know! the reconnect works in a robust way... had never failed on my services...

yashvanthb, 2020-06-09:
reconnection is working the problem with session

AndreasHeine, 2020-06-09:
serverproblem for sure... sound like session timeout to high => session has to timeout faster so server can remove the session!

yashvanthb, 2020-06-09:
if i want to check the connection in every case then the session time will increase?

AndreasHeine, 2020-06-09:
most servers have a session timeout default of 3600 s = 1 hour => if a connection breaks, for example the client dies the server stores the session for 1 hour before deleting it... if client dies he cant resume to the session... the problem with that is your number of serverside sessions is limited and in 1 hour alot can happen... so the client makes a new session => server session +=1 if you repeat till server session limit => error!

yashvanthb, 2020-06-09:
how will come to know the session time out from server in general

AndreasHeine, 2020-06-09:
prosys... dont know!? Sorry.

yashvanthb, 2020-06-10:
not in prosys in general server