FreeOpcUa/python-opcua #1069: Cannot write tag value to Siemens S7-1200/1500 PLC

gbilodeau007, 2020-06-09:
I have a strange issue with not being able to write a tag value.  I did get a Bool to work, but no int/floats ?  No problem reading them and I can write them with ‘Integration Objects’ OPC UA Client program.  I have tried on Linux PI-4, Windows 10 Home & Professional.  I changed the format many times trying what I could find on the web, but always the same error.

error = opcua.ua.uaerrors._auto.BadWriteNotSupported:  "The server does not suppot writing the combination of value, Status and timestamps provided.

CODE:

OPC Demo 
#
#	S7-1500 & Windows10 machine
#	6/8/2020
#

from datetime import datetime, time
import sys
from opcua import Client
from opcua import ua
from array import *
import sqlite3
from sqlite3 import Error
import numpy as np
from opcua import Node
from opcua import crypto
from opcua.tools import endpoint_to_strings
print ("Import Done OK")

# client = Client("URL") #connect using a user to the PLC OPC UA Server
url = "opc.tcp://192.168.3.1:4840"
myclient = Client(url)

myclient.connect()
print ("Connect Done")
root = myclient.get_root_node()
print("Root Node", root)
print("Children of root are: ", root.get_children())

#dv.ServerTimestamp = datetime.now()
#dv.SourceTimestamp = datetime.now()

# Now
tagpath = 'ns=3;s="BIT_TAG"'
myvar = myclient.get_node(tagpath)
obj = myclient.get_objects_node()
val_Bool = myvar.get_value()
print("myvar is: ", myvar)
print("myobj is: ", obj)
print("my Bool Value is", val_Bool)
myvar.set_attribute(ua.AttributeIds.Value, ua.DataValue(True))
print(' ')

tagpath = 'ns=3;s="WORD_TAG"'
myvar = myclient.get_node(tagpath)
obj = myclient.get_objects_node()
val_Word = myvar.get_value()
print("myvar is: ", myvar)
print("myobj is: ", obj)
print("my Word Value is", val_Word)
print(" Data Type ")
myvar.set_value(948, ua.VariantType.Int16)
print("  ")

tagpath = 'ns=3;s="REAL_TAG"'
myvar = myclient.get_node(tagpath)
obj = myclient.get_objects_node()
val_Real = myvar.get_value()
print("myvar is: ", myvar)
print("myobj is: ", obj)
print("my Real Value is", val_Real)
myvar.set_attribute(ua.AttributeIds.Value, ua.DataValue(94.8))
---------------------------------------------------------------------------------------------
error = opcua.ua.uaerrors._auto.BadWriteNotSupported:  "The server does not suppot writing the combination of value, Status and timestamps provided.

AndreasHeine, 2020-06-09:
the way you set the value is wrong!

jake1026, 2020-10-06:
> the way you set the value is wrong!
> 
> try:
> `node.set_value(ua.DataValue(ua.Variant(5.0, ua.VariantType.Float)))`
> 
> hint:
> https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/common/node.py
> 
> ```
>     def set_value(self, value, varianttype=None):
>         """
>         Set value of a node. Only variables(properties) have values.
>         An exception will be generated for other node types.
>         value argument is either:
>         * a python built-in type, converted to opc-ua
>         optionnaly using the variantype argument.
>         * a ua.Variant, varianttype is then ignored
>         * a ua.DataValue, you then have full control over data send to server
>         WARNING: On server side, ref to object is directly saved in our UA db, if this is a mutable object
>         and you modfy it afterward, then the object in db will be modified without any
>         data change event generated
>         """
>         datavalue = None
>         if isinstance(value, ua.DataValue):
>             datavalue = value
>         elif isinstance(value, ua.Variant):
>             datavalue = ua.DataValue(value)
>             datavalue.SourceTimestamp = datetime.utcnow()
>         else:
>             datavalue = ua.DataValue(ua.Variant(value, varianttype))
>             datavalue.SourceTimestamp = datetime.utcnow()
>         self.set_attribute(ua.AttributeIds.Value, datavalue)
> ```

Hello Andreas,
I am glad to have the opportunity to meet you here. First of all, I have followed you. Thank you for your many answers.
I also got problems when I try to use "client.set_values", the error is 
"error = opcua.ua.uaerrors._auto.BadWriteNotSupported: "The server does not suppot writing the combination of value, Status and timestamps provided."
Can you give me an example of this method? Because if you write the values one by one, the speed will become very slow.
Thank u...Jake

AndreasHeine, 2020-10-06:
@jake1026 this error typically means that you dont provide a datavalue like `node.set_value(ua.DataValue(ua.Variant(5.0, ua.VariantType.Float)))`

> "error = opcua.ua.uaerrors._auto.BadWriteNotSupported: "The server does not suppot writing the combination of value, Status and timestamps provided."

Datavalue: "A value with an associated timestamp, and quality."
https://github.com/FreeOpcUa/python-opcua/blob/a7f21f738d805c25fa25241e94225ba35b337dab/opcua/ua/uatypes.py#L857
Variant: A value which can be an Array defined by Valuerank and Arraydimensions
https://github.com/FreeOpcUa/python-opcua/blob/a7f21f738d805c25fa25241e94225ba35b337dab/opcua/ua/uatypes.py#L683

> Because if you write the values one by one, the speed will become very slow.

i guess its not related to the ua-error if your writing performance is not fast enough you should try to restructure your data into arrays and custom-structures even distribution via opcua method would be an option but its all up to the usecase!

jake1026, 2020-10-06:
Thank you for your fast feedback. Do you know why we can not use "client.set_values(NodeList,FloatList)" to write the value to PLC, we just can use "client.set_values(NodeList,DataValueList)" to write value.
Looking forward to ur reply.

AndreasHeine, 2020-10-06:
looks like the convertion from value to datavalue has a bug...
```python
    def set_values(self, nodes, values):
        """
        Write values to multiple nodes in one ua call
        """
        nodeids = [node.nodeid for node in nodes]
        dvs = [ua_utils.value_to_datavalue(val) for val in values]
        results = self.uaclient.set_attributes(nodeids, dvs, ua.AttributeIds.Value)
        for result in results:
            result.check()
```
the problem is here:

> datavalue = ua.DataValue(ua.Variant(val, varianttype))

the variant type is not specified (optional argument)
> dvs = [ua_utils.value_to_datavalue(val) for val in values]

```python
def value_to_datavalue(val, varianttype=None):
    """
    convert anyting to a DataValue using varianttype
    """
    datavalue = None
    if isinstance(val, ua.DataValue):
        datavalue = val
    elif isinstance(val, ua.Variant):
        datavalue = ua.DataValue(val)
        datavalue.SourceTimestamp = datetime.utcnow()
    else:
        datavalue = ua.DataValue(ua.Variant(val, varianttype))
        datavalue.SourceTimestamp = datetime.utcnow()
    return datavalue

```

the result is python-opcua has to guess the uadatatype which is in some cases impossible to get right (value=+5 can be Int or UInt and 16/32bit long...)

https://github.com/FreeOpcUa/python-opcua/blob/a7f21f738d805c25fa25241e94225ba35b337dab/opcua/ua/uatypes.py#L683

> if not variant type is given, attemps to guess type from python type

i reccomend all to always define varianttypes -> python is a dynamic typed language and opcua is static typed!

AndreasHeine, 2020-10-06:
maybe its possible to pass a variant list in addition to nodes and values! i'll give it a try hang on :)

AndreasHeine, 2020-10-06:
maybe something like:
```
    def set_variants(self, nodes, variants):
        """
        Write variants to multiple nodes in one ua call
        """
        nodeids = [node.nodeid for node in nodes]
        dvs = [ua_utils.value_to_datavalue(var) for var in variants]
        results = self.uaclient.set_attributes(nodeids, dvs, ua.AttributeIds.Value)
        for result in results:
            result.check()
```

jake1026, 2020-10-06:
Thank you very much for China...Jake

closed 2020-10-06 by AndreasHeine