FreeOpcUa/python-opcua #1070: Subscript Error While Communicating with Copa Data Process Gateway

Dhruva0707, 2020-06-13:
I have been attempting to subscribe to nodes from a Zenon Process Gateway opcua server. 
The thing is, I can use the get_value() method to collect data from the said node, however, if I subscribe, I get a warning suggesting a Time out in the subscription.

NOTE: I cannot use get_value as I have to keep polling thousands of tags simultaneously. So subscription is prefered. 

The data is changing, at about 1-2 times/sec as I observed from a third-party client (UA-Expert). Any help is welcome. 

 ```python
# import opcua_configuration_apis as srv
# import opcua_driver as drv
from time import sleep
from opcua import Client
from logging import DEBUG

# url = "opc.tcp://localhost:4881"
url = "opc.tcp://localhost:4841"

client = Client(url)
# client = Client(url)

client.connect()
# client2.connect()

client.uaclient._uasocket.logger.setLevel(DEBUG)

# client1.uaclient._uasocket.logger.setLevel(50)
# client2.uaclient._uasocket.logger.setLevel(50)

# nodeId1 = "ns=1;i=2002"
nodeId = "ns=2;s=Logic/Global/L01M07_Availability"

class SubHandler():
    """
        More advanced subscription client using Future, so we can wait for events in tests
        """

    def __init__(self):
        print("INITIALISED SUB HANDLER")

    def reset(self):
        pass

    def datachange_notification(self, node, val, data):
        print("\n\n\n********** DATA CHANGE *******************\n\n\n")
        node_val = val
        print("node val =", node_val)

    def event_notification(self, event):
        print("\n\n\n********** EVENT *******************\n\n\n")

node = client.get_node(nodeId)
print("node: ", node)
myhandler = SubHandler()

node_val = node.get_value()
print("node val = ", node_val)

sub = client.create_subscription(500, myhandler)
print("Subscription: ", sub)
handle1 = sub.subscribe_data_change(node)

while True:
    try:

        sleep(1)

    except KeyboardInterrupt:

        # client1.disconnect()
        client.disconnect()
        exit()

```

Output: 

(OPCUA) C:\Users\itanta\OPCUA>python test.py
node:  Node(StringNodeId(ns=2;s=Logic/Global/L01M07_Availability))
INITIALISED SUB HANDLER
node val =  76.6812973022461
Subscription:  <opcua.common.subscription.Subscription object at 0x0000007D95EA8D08>
ServiceFault from server received while waiting for publish response
ServiceFault from server received while waiting for publish response
ServiceFault from server received while waiting for publish response
ServiceFault from server received while waiting for publish response
ServiceFault from server received while waiting for publish response
ServiceFault from server received while waiting for publish response
ServiceFault from server received while waiting for publish response
ServiceFault from server received while waiting for publish response
.
.
ServiceFault from server received while waiting for publish response
Received data from unknown subscription: 0

(OPCUA) C:\Users\itanta\OPCUA>

AndreasHeine, 2020-06-13:
a wireshark trace would be helpfull, code looks right! But you are getting an error from the server back. Means: your asking a publish request end you got no publish response... What is that Server exacly?

Dhruva0707, 2020-06-13:
> a wireshark trace would be helpfull, code looks right! But you are getting an error from the server back. Means: your asking a publish request end you got no publish response... What is that Server exacly?

Thank you for the response. It is a zenon server, with a simulation running in the background (to update the said values. I can browse the server and the said node was found accordingly. Well the thing is a subscription can be created to it from UaExpert (or perhaps it doesn't wait for a response there?).

I have never used Wire Shark before. I will check it up. Any tips?

AndreasHeine, 2020-06-13:
choose the right network interface and filter opcua protokoll

Dhruva0707, 2020-06-13:
Will do!

Dhruva0707, 2020-06-13:
It is on a remote server, I am waiting for a permission from the admin to restart after wireshark installation

Dhruva0707, 2020-06-14:
I found a difference between the connection process from UAexpert and the python code. 
It appears Uaexpert interacts in the following way:
1> SubscriptionRequest -> Server Replies
2> CreateMonotoredItemsRequest -> Server Replies
3> PublishRequest -> Response 

and continues

While, For python client, the sequence is some what like the following:
1> SubscriptionRequest -> Server Replies
2> PublishRequest -> Server Return **BadTimeOut**
3> CreateMonitoredItemsRequest -> Server Replies
4> PublishRequest -> ServerReturns **BadTimeOut**
.
.

And it continues this way

AndreasHeine, 2020-06-14:
well i dont know anything about the server... 

http://download.copadata.com/fileadmin/user_upload/Downloads/Dokumentation/711SP0/GERMAN/Handbuch/Process_Gateway.pdf

should work:
[image]

anything in the server logs?

oroulet, 2020-06-15:
@Dhruva0707 strange. nice if you can send a wireshark log with python-opcua and one with uaexpert doing the same simple thing so we can try to spot any difference

Dhruva0707, 2020-06-15:
I checked, on wireshark on a publish request, it returns a time out error. I am playing around with the library to make all the requests seem like the ones on UaExpert. Once that is done, I will update you. 

The server logs detected no error. Also I am a new to zenon, so nothing I can say for certain. I went in warning and info moods too, still nothing from the process_gateway.

Dhruva0707, 2020-06-16:
Hi,
I resolved the issue. 
the  **publish** function in **ua_client** sends a request with _TimeoutHint_ = 0. 
I understand that ideally this means no timeout boundary, however the zenon server cannot understand this and gives a **BadTimeout**  serverfault.

All I did was change the default timeout to the max that is possible in 2 bytes. 

Thanks a lot for the help!
Do you think I should branch and give the timeout option for the user? (one where it is used for the publish request ass well? )

swamper123, 2020-06-16:
I think this wouldn't be neccessary, but good to know that one mystery is solved and listed up here in the issues. I think we can close that here.

closed 2020-06-16 by Dhruva0707

zerox1212, 2020-06-16:
If anything you should notify the developers of Zenon server that they have a bug.

Dhruva0707, 2020-06-17:
> 
> 
> If anything you should notify the developers of Zenon server that they have a bug.

I agree. this may be an issue with many servers though. Especially products like process gateway based on older Scadas.