FreeOpcUa/python-opcua #1071: missing f-string or .format() at some places

AndreasHeine, 2020-06-14:
while working on asyncua PR #211 i saw some of these and in python-opcua aswell:

```
def _to_val(objs, attr, val):
    from opcua import ua
    cls = getattr(ua, objs[0])
    for o in objs[1:]:
        cls = getattr(ua, _get_uatype_name(cls, o))
    if cls == ua.NodeId:
        return "NodeId.from_string('val')"  
    return ua_type_to_python(val, _get_uatype_name(cls, attr))
```

if it does not return a f-string it is unable to pass val into the string!?

@oroulet : ???

swamper123, 2020-06-15:
Where is that snippet located?

AndreasHeine, 2020-06-15:
schemas / generate address space

swamper123, 2020-06-15:
This is correct.
I will have a look in general over other lines, where f-string could be added (for performance) and possible typos like this one.

oroulet, 2020-06-15:
yes something worng in code over. IT is strange there it is not caught by any tests. Is it such a special case?

swamper123, 2020-06-15:
possible .format replacements with f-strings:
- generate_ids.py
- download.py
- generate_address_spaces.py
- generate_events_objects.py
- generate_model.py
- generate_protocol_python.py
- generate_statuscode.py
- uaprotocol_auto_add.py

possible typos?
- generate_address_space.py, line 150
- generate_event_objects.py, line 47

others haven't been found at the moment.

oroulet, 2020-06-15:
.format still work in python  3. but it is nice to fix anyway
Concerning the typos can you send link to code on github? the line number is not easy to use with all these different branches...

swamper123, 2020-06-15:
The first:
https://github.com/FreeOpcUa/python-opcua/blob/f761ccf0882daff8f000885e851db66b3c0ea167/schemas/generate_address_space.py#L150

and the second one:
https://github.com/FreeOpcUa/python-opcua/blob/f761ccf0882daff8f000885e851db66b3c0ea167/schemas/generate_event_objects.py#L47

On the first sight they seem to be a similar problem, but I could be wrong.
The format stuff was just for possible performance stuff, if we are touching some strings in that case anyway.

oroulet, 2020-06-15:
I am not sure at all these are typos. This is code written for code generation and somethings it looks quite strange. Maybe you have a varaible called "message" in the generated code.
The first one looks correct because there is for sure a variable called ua.ObjectIds.String in the generated code.

oroulet, 2020-06-15:
but that line `return "NodeId.from_string('val')"`is suspicious because even if there is a variable called val in code, then it is written as a string...

AndreasHeine, 2020-06-15:
```python
    if cls == ua.NodeId:
        return f"NodeId.from_string('{val}')"
```

this is never called at all...

closed 2020-06-16 by AndreasHeine