FreeOpcUa/python-opcua #1072: Subscription Error: Stop getting data change values after 2 mins

DionisisBozikis, 2020-06-15:
Hello,
I am having the problem described in the title. Basically if none of the monitored node iDs changes its value for longer than 2 minutes, after that the Subscription script stops working.
Am i doing something wrong?
i get no errors. if i restart the script it works normaly, catches thelast data change, and if it idle again for longer than 2 minutes, it stops working.

here is my code:
`import time
from opcua import Client
from opcua import ua
from opcua import Subscription

start=time.time()
plc_url = "opc.tcp://192.168.2.201:4840"
production_status_node_id = "ns=4;s=|var|CPX-E-CEC-M1-PN.Application.SIMULATED_VARIABLES_PLC.GVL.PRODUCTION_STATUS"
trigger_node_id_string = "ns=4;s=|var|CPX-E-CEC-M1-PN.Application.SIMULATED_VARIABLES_PLC.GVL.TRIGGER_SIMULATION"

class SubHandler(object):
    """
    Subscription Handler. To receive events from server for a subscription.
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another
    thread if you need to do such a thing
    """
    def datachange_notification(self, node, val, data):
        print(node,":", val)

#=============opcua==============#
client = Client(plc_url)
client.connect()
#===============================#

production_status_node_object = client.get_node(production_status_node_id)
trigger_node_object = client.get_node(trigger_node_id_string)

production_handler = SubHandler()
sub_production_handler = client.create_subscription(200, production_handler)
sub_production_handler.subscribe_data_change(trigger_node_object)

while True:
    print(round((time.time()-start)/60, 2))
    time.sleep(15)`

DionisisBozikis, 2020-06-15:
i have found a walk-around by making a daemon thread that keeps alive the connection with polling. just a get_value() every 1 minute.
But is there a more "pretty" solution ??

AndreasHeine, 2020-06-15:
you should of course polle a node for connection monitoring but thats a different topic... typicaly in a subscription there should be an publish request from your client and a following publish response from your opcua server (you can check it with wireshark + filter opcua protokoll) most of the time you see then whats going on and when it breaks! any other way is just guessing... code looks ok!

DionisisBozikis, 2020-06-15:
i eventually subscribed to a watch dog counter that changes every now and then to keep alive the connection. i will keep the issue up for some days in case someone has any idea and then i will close it.
Thanks you very much for your quick response!

brubbel, 2020-06-20:
More info is needed.
- Which server is used?
- Connection directly via LAN? Some NAT's tend to close connection when there is no traffic for some time.
- Does the disconnects also happen when you subscribe to the server's ServerStatus->CurrentTime node?
- Are you using encryption to connect? There were/are some issues with the refresh of encryption tokens. Try again without encryption.