FreeOpcUa/python-opcua #1077: Client accepting the server certificate

gilbertsmink, 2020-06-23:
Hi everyone,

Is it possible for the client to set the security to Basic256 and simply accepting the server certificate? Thus, without using any client certificate and private key?

This seems possible in UA expert, or am I missing something?

Thanks in advance

AndreasHeine, 2020-06-23:
This is not how SSL/TLS secure communication work if you mean message security!
or do you mean Authentification?
Your question is kind of confusing...

gilbertsmink, 2020-06-23:
Sorry that my question is not clear.

Let me try to explain again better. I have two OPC UA servers I can connect to, both allow Anonymous authentication, thus no username/password or certs. But, they are different in their security policy (hope this is the right term). The first server allows None security policy, the second does not. 

I can connect to the first server, but can't figure out how to connect to the second, using Basic256. 

[image]

So to conclude, to me it seems UA Export allows for using Basic256 without certs, or am I missing something?

AndreasHeine, 2020-06-23:
UaExpert generates certs while installing... if you want to use just "Sign" as message security mode you need something to sign with => cert (and a key for generating the cert)

from my point you should always use Basic256Sha256 Sign&Encrypt if its sensetiv data and always protect your server with auth! 
(i always recommend using username/password auth for the server as minimum especially if its a PLC where you can change parameters or break things) i personly dont like cert-auth because there self signed (what is a self printed passport worth? only paper and ink!)

gilbertsmink, 2020-06-23:
Hi Andreas, thanks for your recommendations. We will indeed use username/password in production. 

so how can i implement Basic256Sha256 in my client without certs, but with username/password?

AndreasHeine, 2020-06-23:
any kind of message security (except "None") needs a cert an a key!

AndreasHeine, 2020-06-23:
you can use "None" + "Username" means your data is not encyptet and not signed but in order to connect you need the login! if you are in a company network where no external people can access it it would be enought