FreeOpcUa/python-opcua #1078: OPC Foundation nodesets are no longer avalible ... !

swamper123, 2020-06-24:
https://github.com/OPCFoundation/UA-Nodeset.git
Mentioned by @AndreasHeine

swamper123, 2020-06-24:
https://github.com/OPCFoundation/UA-Nodeset/issues/64#issue-643861345 ---> Leave a thumb up

oroulet, 2020-06-24:
and these people say they work for interoperability...

zerox1212, 2020-06-24:
I suppose open source community will need to start maintaining our own nodeset...

AndreasHeine, 2020-06-24:
luckly i regenerate befor that! still have latest master from two weeks ago :)

AndreasHeine, 2020-06-24:
https://github.com/AndreasHeine/ua-nodesets

swamper123, 2020-06-25:
But it still will be a lot of "fun" to generate them for future changes...anyway I am inquisite to know their explanation.

AndreasHeine, 2020-06-25:
the core of opcua is a open plattform! all have the same base (xml, csv, ....) its in my opinion not helpfull with double data (room for errors)... hope they change there minds and at least give us always the latest release.

oroulet, 2020-06-25:
Are they still available on opc website?

swamper123, 2020-06-25:
They seem to be still available.

AndreasHeine, 2020-06-25:
> Instructions on requesting access to the member only repository can be found here.

I am member in the foundtion but my github account has a different mail.... so i need to setup a second account!? to get the latest nodesets. long things short we will not run dry but the way we get the files is not convenient anymore...

swamper123, 2020-06-25:
@AndreasHeine So you are Corporate (or greater) member? I guess the e-mail thing would be possible after a long while, lots of email etc., but before you may "PR a selfmade Nodeset update" you should study the licence. Not that you get yourself in toruble in any kind of way.

AndreasHeine, 2020-06-25:
some weeks ago i joint a VDMA OPC Joint Working Group (companion spec.) maybe the VDMA ia a way to get access to more information!

swamper123, 2020-06-25:
For all these, who may not know why this is important and what is happening here:

The foundation released different Nodesets on their github account. These Nodesets are "blueprints" of how the server nodes has to be build up. If you have a look in [here](https://github.com/FreeOpcUa/opcua-asyncio/tree/master/asyncua/server/standard_address_space) you see how "small" these nodesets are.
So if the Foundation is making changes, we wouldn't be able to provide a compatible Version of an OPC UA Server for these newer versions. It is not clear (for me at the moment) if we would be allowed to use them, if an OPC corporate member would pass them through (even with some changes), but it would be sillym because we would need these Nodesets 1:1.
Other aspects are mentioned by in the post at the top.
(In case anything is faulty in this post or needed to be edit, feel free to comment below).

AndreasHeine, 2020-06-25:
in addition all released companion spec. nodesets where included aswell!

AndreasHeine, 2020-06-25:
Yes i am! Yesterday was my first JWG Meeting :)

AndreasHeine, 2020-06-28:
there are back and avalible! 😊🎉