FreeOpcUa/python-opcua #1080: Subscription concept Server

nikiFoe, 2020-06-29:
I have the problem understanding the subscription concept in FreeOpcUa. 
At first, I thought it is mandatory to subscribe a node on the server side like this:
```
handler = SubHandler()
sub = server.create_subscription(10000, handler)
handle = sub.subscribe_data_change(myvar)
```

This works as far as publishing in the "period" that is in this case 10000ms. Problematic, I also can also publish the value of my node with:

`myvar.set_value(v)`

Evertime  I set the value it is atomatically published to the client. This troubles my because I want to change the value `v` in a different thread like this: 

```
class VarUpdater(Thread):
     def __init__(self, var):
        Thread.__init__(self)
        self._stopev = False
        self.var = var
        self.val = 0

    def stop(self):
        self._stopev = True

    def run(self):
        while not self._stopev:
            v = sin(time.time() / 10)
            self.var.set_value(v)
            time.sleep(2)
```

If I monitor my `myvar` like discribed in the begining with the `Subhandler()` and change the value in the thread, my Client recives both, the notification from the subscription and the `set_value(v)` which makes the value pubshished two times . 
Is there a way to change the value of `myvar` whithout automatically publishing it?

AndreasHeine, 2020-06-29:
> I have the problem understanding the subscription concept in FreeOpcUa.

First of all the subscription we are talking about is the monitored item service set: https://documentation.unified-automation.com/uasdkhp/1.0.0/html/subscription_model.png

> Is there a way to change the value of myvar whithout automatically publishing it?

the behavior that a value change triggers a publish on a data change sub is ok as it is... maybe you take a look on [asyncua](https://github.com/FreeOpcUa/opcua-asyncio) guess threads some times tricky and asyncio is makes the code much simpler without taking care on the hole threading starting stopping stuff!

but maybe i got it wrong what kind of behaivior did you expect or what are you trying to get done just a value updater which updates a value each 10s ?

nikiFoe, 2020-06-29:
Hey Andreas,

I will probably change the thread to the asyncua stuff. 
But you basically mean that, for instance,  a node that is subscribed over the client  already has automatically a event/datachange trigger on the server?  Does this mean when I subscibe over the client to a node I don't need to implement a specific subscription like so on my server side?:

```
handler = SubHandler()
sub = server.create_subscription(10000, handler)
handle = sub.subscribe_data_change(myvar)
```

If this is right what purpose does the subsciption on the server side can have?

Thanks for your answere

AndreasHeine, 2020-06-29:
the client which subscribed to the nodes datachange event will be notified so you dont need to implement anything serverside! serverside subscription you can use to get data internaly so you dont need to read cyclic... with an data change event you could change server values or node values in an event/callback way

nikiFoe, 2020-06-29:
Ok, perfekt. Now I understand.
Thank you.

closed 2020-06-29 by nikiFoe

zerox1212, 2020-06-30:
@nikiFoe Just as an example the server history uses internal server-side subscriptions (for better or worse...) to record data changes to database.