FreeOpcUa/python-opcua #1082: required argument is not an integer

chandlergee, 2020-07-01:
[image]

chandlergee, 2020-07-01:
```python
import sys
sys.path.insert(0, "..")
import logging
import time

from opcua import Client
from opcua import ua

class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)

    client = Client("opc.tcp://118.24.36.220:62547/DataAccessServer")
    try:
        client.connect()
        client.load_type_definitions() 
        handler = SubHandler()
        ids = ['ns=2;s=Machines/Machine A/IsFault']
        myvar = [client.get_node(x) for x in ids]
        sub = client.create_subscription(500, handler)
        handle = sub.subscribe_data_change(myvar)
        time.sleep(0.1)
        sub.subscribe_events()
        
        input("hello")

        sub.unsubscribe(handle)
        sub.delete()
        input("hello")

    finally:
        client.disconnect()
```

chandlergee, 2020-07-01:
I have a test,the function delete is ok

AndreasHeine, 2020-07-01:
what is type of handle?

chandlergee, 2020-07-02:
a list with int elements

AndreasHeine, 2020-07-02:
then you need to loop over it and unsub each!

swamper123, 2020-07-02:
If I can sub to many nodes, wouldn't it be intuitive to unsub them all as well? Because a handle should handle the stuff I ordered (like sub to many nodes).
Or could there be a pitfall in any kind of way?

AndreasHeine, 2020-07-02:
we could make it case sensetiv if arg is list iterate over it!

`If I can sub to many nodes, wouldn't it be intuitive to unsub them all as well? `
guess sub.delete() should be the right way

swamper123, 2020-07-02:
[Just had a look at opcua-asyncio where this was already implemented.](https://github.com/FreeOpcUa/opcua-asyncio/blob/88bbc425eb2137917795640196fbde8712898681/asyncua/common/subscription.py#L292). So I guess this can be closed, since this repo is in maintainance,

AndreasHeine, 2020-07-02:
even then you can make a pr if you like maybe its just copy paste and remove await!?

swamper123, 2020-07-02:
I will drop one in a minute

PR #1085 (2020-07-02): Add list handeling for unsubbing
should fix #1082
closed 2020-07-02 by oroulet