FreeOpcUa/python-opcua #1086: BaseEventLoop._create_server_getaddrinfo

Jayant-Pancholi, 2020-07-02:
I have Initialized all async functions in a class, Calling them main using asyncio.run(func()).
Getting error for Server Start ()

RuntimeError: Task <Task pending name='Task-3' coro=<ServerAsyncUA_0207.StartServer() running at D:/WORKSPACE/OPC/modbus_opcua_SINGLEMAC_Setup/opcua_server/scratch_8.py:131> cb=[_run_until_complete_cb() at C:\Users\jayant.pancholi\AppData\Local\Programs\Python\Python38-32\lib\asyncio\base_events.py:184]> got Future <_GatheringFuture pending> attached to a different loop
sys:1: RuntimeWarning: coroutine 'BaseEventLoop._create_server_getaddrinfo' was never awaited

Calling it as like -

if __name__ == '__main__':
    main()

def main():
    SAO = ServerAsyncUA_0207()

asyncio.run(SAO.StartServer())

async def StartServer(self):
     async with self.server:
            while 1:
                for each in self.var_list:
                    await asyncio.sleep(0.1)
                    await each.set_value(random.randint(1, 100))

swamper123, 2020-07-02:
I think you may want to use [opcua-asyncio](https://github.com/FreeOpcUa/opcua-asyncio) instead of python-opcua.

swamper123, 2020-07-02:
And I am a bit confused what ServerAsyncUA_0207 is. Have you used an alias import?

Jayant-Pancholi, 2020-07-02:
Thanks for the input, But I am using asyncio and asyncua to implement server.

Jayant-Pancholi, 2020-07-02:
ServerAsyncUA_0207 is a Class name in which I have various async defined functions.

swamper123, 2020-07-02:
Okay. Well your error message says, that 'BaseEventLoop._create_server_getaddrinfo' is missing an await.

Jayant-Pancholi, 2020-07-02:
But that is coming from lib\asyncio\base_events.py this path. Library specific.
what is that It is looking for.
Is it like there are multiple async used.

Error also reads as - 

RuntimeError: Task <Task pending name='Task-3' coro=<ServerAsyncUA_0207.StartServer() running at D:/WORKSPACE/OPC/modbus_opcua_SINGLEMAC_Setup/opcua_server/scratch_8.py:131> cb=[_run_until_complete_cb() at C:\Users\jayant.pancholi\AppData\Local\Programs\Python\Python38-32\lib\asyncio\base_events.py:184]> got Future <_GatheringFuture pending> attached to a different loop

swamper123, 2020-07-02:
tbh I still don't know what your issue has to do with this repo(python-opcua), especially when [opcua-asyncio](https://github.com/FreeOpcUa/opcua-asyncio) is the opc server which uses async stuff.

Because I don't know your code, I can just tell you, that a future object/coroutine is missing an await.

Also I can say, that you may have a look at [this](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-example.py) example.

in short:

```
async def main():
    [...]
    server = Server()
    await server.init()
    [...]
    async with server:
        [...]
```

Jayant-Pancholi, 2020-07-02:
how does "async with server" works ?

swamper123, 2020-07-02:
`async with` is new since python 3.6 I guess. Here the [link to the docs](https://docs.python.org/3/reference/compound_stmts.html?highlight=async%20await), at the bottom.
You need two more private methods in your server class: `__aenter__` and `__aexit__`,
If you enter an async statement, like `async with server:`,` __aenter__` will be triggered, and when you get out of it, `__aexit__`.

[You can see it here](https://github.com/FreeOpcUa/opcua-asyncio/blob/4f83bad85ff32c38ea774aeae8eb1ed40fdaacf8/asyncua/server/server.py#L148) how they implemented it.