FreeOpcUa/python-opcua #1089: Regarding HDA Data and Timestamp values viewed in UA Expert Client does not match with the PYTHON OPCUA Client Response

rajnishnationfirst, 2020-07-03:
Hi ,

I have developed a PYTHON OPCUA client to get data from Siemens WinCC SCADA .

**The above Python OPCUA client fetched data for a TAG ( Archived ) for the below snapshot.**

[image]

I queried using the above time and gets the data in the form of list having value , status and timestamp.

Also i used the UA Expert OPCUA client to view the same interval HDA data .

But the values doesn't gets mached against the timestamp values .

**Response of Python OPCUA HDA cleint :**
[image]

**The same interval data viewed from UA expert doesn;t match .**

[image]

Also is there any other OPC UA HDA Software based client to view the OPCUA HDA data .

Help from anyone from this group will be really appreciable for me to learn something new.

Thanks
Rajnish Vishwakarma

brubbel, 2020-07-05:
It seems there is an off-by-one shift between source timestamp and values in one of the two clients.
That or the server does not send consistent results in those two history requests.

Most interesting, it would be useful to find which is the origin of that error.

duduyoyo, 2024-02-01:
See if [this](https://github.com/duduyoyo/WebSocket4OPC) can help you.