FreeOpcUa/python-opcua #1090: Get id of a node as the value of i and ns

ohidurbappy, 2020-07-05:
In my subscription handler I want to get the id  of node. More specifically check what's the value of i and ns of a node id...

class SubHandler():
    def datachange_notification(self, node, val, data):
        print(datetime.now(), "- Data Change - ", val)
        i=node........
        ns=.......

    def event_notification(self, event):
        pass

AndreasHeine, 2020-07-05:
> class SubHandler():
> def datachange_notification(self, node, val, data):
> print(datetime.now(), "- Data Change - ", val)
> i=node........
> ns=.......

since last release (0.98.12):
`str(node)` => returns a string like this "ns=4;i=1000"
then you could split in ";" then "=" or use a regex to get ns.

ohidurbappy, 2020-07-05:
Thank you. It solves the problem.

closed 2020-07-05 by ohidurbappy

brubbel, 2020-07-05:
```
i = node.nodeid.Identifier
ns = node.nodeid.NamespaceIndex
```

ohidurbappy, 2020-07-05:
> ```
> i = node.nodeid.Identifier
> ns = node.nodeid.NamespaceIndex
> ```

This is more neat solution.

brubbel, 2020-07-05:
It is useful to keep in mind that Node is just some object wrapper around the NodeId + a server session, providing some useful tools to interact (e.g. get_value, get_browsename, ...) with the client proxy or the server.

pseudocode:
```
class Node:

    def __init__():
        self.nodeid
        self.session

   def useful_tools():
       ...
```

AndreasHeine, 2020-07-05:
> i = node.nodeid.Identifier
> ns = node.nodeid.NamespaceIndex

good to know! havn't seen that, in the past i only used "ns=4;i=1000".