FreeOpcUa/python-opcua #1091: Data Exported from Siemens WinCC SCADA in Table format and viewed in UAExpert client doesnot match with data fetched with node_id.read_raw_history(starttime,endtime,0)

rajnishnationfirst, 2020-07-05:
Hi Everyone ,

I fetched a **archived** tag data  using **node_id.read_raw_history(starttime,endtime,0)** it feches 122 records having 
 0.5 second (or  **500 Microsecond**) and it is fecthed successfully .

[image]

**Also when i viewed in a table data ( feature of WinCC Siemens SCADA ) i get the data as below format** :

[image]

**At the same i viewd the data in UA Expert and i viewed the data as attched below** 👍 
[image]

Issues i am facing : 

The data in UA Expert and the TABLE VIEW ( Feature of SIEMENS WInCC SCADA ) is matching , while the data recieved via read_raw_history doesnot matched ( which some matches as seen in the below attachment .

[image]

What may be the issue in **node_id.read_raw_history(starttime,endtime,0)**?

How can i get the correct data using **node_id.read_raw_history(starttime,endtime,0)** or i will have to use some othr functions like **node_id.history_read(params)**

It would be great if anyone can assist me on this and provide me the appropriate reply which will help me to proceed further.

Thanks and Regards
Rajnish Vishwakarma

brubbel, 2020-07-05:
It seems to me that the timestamps are matching (if you shift the first red box 1 row up).
However, the last value at second 48.578 (pdf row 10) is not matching.

rajnishnationfirst, 2020-07-05:
Thanks brubbel for your kind reply.
I have now changed the range of UA Expert and snapped it for you to view .
Now you can pls check that the UA Expert & Table VIEW ( feature of SIEMENS WInCC SCADA ) is same and the data rcived snapped using read_raw_history is different ( I mean to say , that UA Expert and Table view has same data and equal but wwhen matched with node_id.read_raw_history(starttime,endtime,0 ) does not matched.

**Pls have a look to all 3 snaps :**
**UA Expert View ( range corrected )**

[image]

PDF View and Data View : 

[image]

Also lastly response snap from node_id.read_raw_history(starttime,endtime,0)

[image]

Yes the red block marked matches with UAExpert & Table View Data ( from SIEMENS WInCC SCADA )

Also if you can help me with some questions 👍 

1) As some data matches (in Red Block )using data fetched using **node_id.read_raw_history(starttime,endtime,0)** , does its a server side issue of issue in **node_id.read_raw_history(starttime,endtime,0)** ?

2) If there is any issue in **node_id.read_raw_history(starttime,endtime,0)** , what is the correct method / function to read and fetched data ?

Thanks and Regards
Rajnish Vishwakarma

zerox1212, 2020-07-06:
From what I remember there is a time conversion in Python for the binary protocol. Native `DateTime` objects are probably converted and maybe some precision is lost. Might be worth digging into what timestamp is coming across the wire and comparing it to what ends up in python DateTime object.

rajnishnationfirst, 2020-07-06:
Thanks Andrew or your reply:
I am quering as given below for a datetime object 

**Code:**
```
start_time = datetime.datetime(2020, 7, 6, 10, 35)
end_time = datetime.datetime(2020, 7, 6,  10,41)

node .read_raw_history(start_time, end_time, 0))
```

Is the above date time object correct?
If wrong , how can i configure datetime object as per your reply you gave above.

Andrew , pls assist me i am stuck here due to this.

Thanks 
Rajnish Vishwakarma

brubbel, 2020-07-08:
I still do not understand why you consistently highlight the column off-by-one?
[image]

zerox1212, 2020-07-08:
I do not think it has to do with your start and end times. It has to due with time conversion that happens here: https://github.com/FreeOpcUa/python-opcua/blob/13b4e1249e06a3d3feef09afd04139eb774207bc/opcua/ua/ua_binary.py#L45

You can investigate this for loss of time precision.