FreeOpcUa/python-opcua #1098: Unable to start with custom url as endpoint

padraic-padraic, 2020-07-09:
Hello,

I am trying to setup an OPC-UA server and I'm not quite clear on the behaviour of `Server.set_endpoint`. If I try to use anything other than `0.0.0.0`, I get an error from asyncio that it cannot bind to the address. For example, using localhost gives

```
server.set_endpoint(`opc.tcp://localhost:4840`) .....
OSError: [Errno 99] error while attempting to bind on address ('::1', 4840, 0, 0): cannot assign requested address
```

And a similar error, with the public IPv4 addres substituted for `::1`, if using the server IP.

Am I not using this correctly?

swamper123, 2020-07-09:
Linux User?

padraic-padraic, 2020-07-09:
Yep, this has happened to me on Pop Os 20.04, Ubuntu 18.04 or in a Docker container based on Debian buster

brubbel, 2020-07-09:
Docker containers do not support ipv6 by default.
https://docs.docker.com/config/daemon/ipv6/

padraic-padraic, 2020-07-09:
My question then I guess is about the resulting endpoints, which have addresses like 172.x.x.x and a discovery url of 0.0.0.0. Would this potentially cause issues when connecting with a client? For example, we're using the National Instruments client and it seems to use the endpointURL to verify the server certificate as part of opening a session.

brubbel, 2020-07-09:
> Would this potentially cause issues when connecting with a client?
Yes.
https://github.com/FreeOpcUa/python-opcua/pull/687

The working solution is replying to the client with the address that is (or should be) included in the client's own discovery request.
This also works over VPN etc.