FreeOpcUa/python-opcua #1103: Read Aggregate data from Wincc OPC UA Server using ReadProcessedDetails function

Prachi10000, 2020-07-15:
Hello,
I am fetching data from wincc opc ua server using history_read function and passing parameters in that as per the code attached below this code is working fine.
Now I want to read aggregated data from using the function ReadProcessedDetails as per given in OPC UA Specification part aggregate part 13 point no 5.2.2 
could you please tell me how i have to pass those parameters (start time , end time, processing Interval,and aggregate type )in this same program .
please if anyone can help me it would be your great help.

`from opcua import Client
from opcua import ua
import datetime
class Siemens_HDA_Client(object):

    def __init__(self):
        print('Constructor Called')

    def get_WinCC_Siemens_HDA_Client_Connection(self):
        client = Client("opc.tcp://EC2AMAZ-VJQV23P:4862")
        return client

def do_transaction_history(client, nodes, start, end, data_per_req, contes=None):
    client.connect()
    if contes is None:
        contes = [b''] * len(nodes)
    details = ua.ReadRawModifiedDetails()
    details.IsReadModified = False
    details.StartTime = start
    details.EndTime = end
    details.NumValuesPerNode = data_per_req
    details.ReturnBounds = False

    params = ua.HistoryReadParameters()
    params.HistoryReadDetails = details
    params.TimestampsToReturn = ua.TimestampsToReturn.Source
    params.ReleaseContinuationPoints = False
    for n, c in zip(nodes, contes):
        valueid = ua.HistoryReadValueId()
        valueid.NodeId = n.nodeid
        valueid.IndexRange = ''
        valueid.ContinuationPoint = c
        params.NodesToRead.append(valueid)
        result = client.uaclient.history_read(params)
    return result

def test_procdata_history(client):
    start = datetime.datetime(2020, 7, 5, 7, 48)
    end = datetime.datetime(2020, 7, 5, 7, 49)
    #start = datetime.datetime(2020, 7, 4, 5, tzinfo=datetime.timezone.utc)
    #end = datetime.datetime(2020, 7, 4, 5, tzinfo=datetime.timezone.utc)
    new_node_1 = client.get_node('ns=1;s=v|HDA_Archives\int')
    new_node_2 = client.get_node('ns=1;s=v|HDA_Archives\float')
    new_node_3 = client.get_node('ns=1;s=v|HDA_Archives\bool')
    nodes = [new_node_1,new_node_2,new_node_3]
    #nodes = [client.get_node(ua.NodeId(33001 + x, 4)) for x in range(10)]
    more_data = True
    cnt = 0
    contes = None
    history_per_request=500
    history_limit=100
    #data_per_req = history_per_request
    data_per_req = history_per_request
    values = []
    while more_data:
        print(data_per_req)
        result = do_transaction_history(client, nodes, start, end, data_per_req, contes)
        print('Result',result)
        cnt += len(result[0].HistoryData.DataValues)
        more_data = False
        values += result[0].HistoryData.DataValues
        more_data = result[0].StatusCode.is_good() and result[0].ContinuationPoint is not None
        if cnt >= history_limit:
            more_data = False
        if history_limit - cnt < history_per_request:
            data_per_req = history_limit - cnt
        if more_data:
            contes = [res.ContinuationPoint for res in result]

p1 = Siemens_HDA_Client()

print('Data')

#p1.get_HDA_Data_From_Siemens_Scada()
test_procdata_history(p1.get_WinCC_Siemens_HDA_Client_Connection())
[image]
`

Prachi10000, 2020-07-15:
i want to replace this ReadRawModified details with ReadProcessedDetails as per below image.

[image]

kawkawa99, 2023-06-20:
I'm facing the same issue at the moment, please if you found the answer can you share it here please ?

duduyoyo, 2024-02-01:
This [solution](https://github.com/duduyoyo/WebSocket4OPC) provides all various HDA read operations, including aggregate. Good luck!