FreeOpcUa/python-opcua #1105: Setting Monitoring mode (Reporting/Disabled) on an item creation request

kaeros, 2020-07-15:
Hello,

I have been working whith the freeopcua python library for quite some time, and before getting to my issue, i wished to thank you again for this open source library.

When I tried to play with monitoring modes (from a client point of view), I found out that there were currently no support for this feature in the code. 
In the file "**/opcua/common/subscription.py**", when the method "**_make_monitored_item_request**" is called, the item is automatically set to the monitoring mode "**Reporting**" (see capture below):

[image]

What I would like to do is to create monitored items with a monitoring mode "disabled". Since the feature exists in the ua library which is imported in freeopcua, it should be possible to forward an argument to the method "**_make_monitored_item_request**" to allow a user to select from this two monitoring modes? 

My question is the following: Does this feature is planned to be activated in the library in next few releases ? (I have not seen any mention about monitoring mode in the roadmap). 
Since it is my first post here, it may be more appropriate to follow this issue in a pull request for this is more of an enhancement request than a prctical issue? If so, I will mark this issue as closed and create the correct ticket.

Best regards,

Mathieu

zerox1212, 2020-07-16:
Why not just use the above code as a template and add your own function with mode "disabled"? I have not seen anyone request this before, so I guess the only way it would get added is if you added this feature and made a PR.

AndreasHeine, 2020-07-29:
@kaeros 

>  it should be possible to forward an argument to the method "_make_monitored_item_request" 

under or dundermethods should never be used from the user it is convention in python that these are "private" or "protected".
sadly its not only done with adding args to "_make_monitored_item_request" you also need to add in -> "_subscribe" -> "subscribe_data_change" and in addition "Disabled" and "Sampling" must be implemented. Aswell as a possibility to modify the monitoring mode (as far as i know "modify_monitored_item" is not designed for that purpose -> 5.12.3 ModifyMonitoredItems -> 5.12.3.2 Parameters -> https://reference.opcfoundation.org/v104/Core/docs/Part4/7.16/)

> My question is the following: Does this feature is planned to be activated in the library in next few releases ? (I have not seen any mention about monitoring mode in the roadmap).

i dont think anyone is working on that... "The library is in maintenance mode." any new changes will be first implemented in [asyncua](https://github.com/FreeOpcUa/opcua-asyncio)

"Disabled": you could simply unsub that item 
"Sampling": sampling without reporting is in my opinion a waste of server resources...

regards Andreas

kaeros, 2020-09-01:
I was abled to implement theses features on my end. I will keep my branch as it is for the time being but I will submit a merge request as soon as I will have have a green light to switch the library from python-opcua to asyncua on my project (since python-opcua library is in maintenance). After then, if the merge request is accepted, the code will probably be also merged in the python-opcua library.

closed 2020-09-01 by kaeros