FreeOpcUa/python-opcua #1107: Not possible to assign all uint64 values to an UInt64 variable

Oidlichtnwoada, 2020-07-21:
I create my variable this way:
[image]
Then I assign the biggest possible value 2 ** 64 -1:
[image]
Then I got the following output:
Listening on 0.0.0.0:4840
Exception in callback InternalSubscription._sub_loop()
handle: <TimerHandle when=499465.468 InternalSubscription._sub_loop()>
Traceback (most recent call last):
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\asyncio\events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\server\internal_subscription.py", line 293, in _sub_loop
    self.publish_results()
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\server\internal_subscription.py", line 321, in publish_results
    self.callback(result)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\server\uaprocessor.py", line 80, in forward_publish_response
    self.send_response(requestdata.requesthdr.RequestHandle, requestdata.seqhdr, response)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\server\uaprocessor.py", line 47, in send_response
    struct_to_binary(response), message_type=msgtype, request_id=seqhdr.RequestId)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 261, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 284, in to_binary
    return struct_to_binary(val)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 261, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 284, in to_binary
    return struct_to_binary(val)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 256, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 296, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 296, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 274, in to_binary
    return pack_uatype(vtype, val)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 188, in pack_uatype
    return extensionobject_to_binary(value)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 460, in extensionobject_to_binary
    Body = struct_to_binary(obj)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 256, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 296, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 296, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 284, in to_binary
    return struct_to_binary(val)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 261, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 274, in to_binary
    return pack_uatype(vtype, val)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 194, in pack_uatype
    return struct_to_binary(value)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 261, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 274, in to_binary
    return pack_uatype(vtype, val)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 192, in pack_uatype
    return variant_to_binary(value)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 378, in variant_to_binary
    b.append(pack_uatype(var.VariantType, var.Value))
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 184, in pack_uatype
    return getattr(Primitives, vtype.name).pack(value)
  File "C:\Users\brantner\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\ua_binary.py", line 135, in pack
    return struct.pack(self.format, data)
struct.error: argument out of range
The values are stored in an int64 field, which does not allow all uint64 values, possibly a special case for uint64 should be added.
The error happens here in ua_binary.py:
[image]
I am looking forward to a response ^^

AndreasHeine, 2020-07-21:
i confirm it for `set_value(18446744073709551615)` but `set_value(ua.DataValue(ua.Variant(18446744073709551615, ua.VariantType.UInt64)))` works!

AndreasHeine, 2020-07-21:
so if you use set_value() you write a python Int64 to the function and it trys to write to a OPCUA UInt64 !!! you can guess what happens ;)

Oidlichtnwoada, 2020-07-21:
Thanks for your VERY fast reply, just as a small comment Python integers are unbounded, there is no bound for integers like long or int in C-like languages in the Python programming language, you can try it out yourself, type in 18446744073709551615 + 1 in the Python console, floats are only bounded to 64 bit in Python because they want to be compliant to IEEE 754.

AndreasHeine, 2020-07-21:
since python 3 long-ints have bin removed... but anyways the problem is set_value() needs more information what to do so pass a Variant or DataValue to it and it should be fine!

closed 2020-07-21 by Oidlichtnwoada

Oidlichtnwoada, 2020-07-21:
The information is already present in the data type of the variable, but if this is the way the framework should work, I am ok with it ^^

AndreasHeine, 2020-07-21:
the opc ua datatype is fix and defined in the server... if you pass a number for example 5 to it, it could be a UInt or an Int you atually can only guess what ua-type it could be, if you have -5 it cant be an UInt but you dont know the bit-length could be an ua-int16, 32, 63. There is some Information present but not enough... all ua stacks in dynamic typed languages have that problem!

Oidlichtnwoada, 2020-07-21:
Try to convert the given data value to the data type defined in the server (using ua.Variant) and if this is not possbible throw an exception, I do not see the problem, you see here

zerox1212, 2020-07-21:
I think we had this discussion before a long time ago. From what I remember the code tries to guess what data type based on the value itself, not what the node defines in the server.

You can read about it here https://github.com/FreeOpcUa/python-opcua/issues/639.