FreeOpcUa/python-opcua #1108: Re-Use of Server instance in another python script

sphemanth, 2020-07-22:
Hello,

I am trying to create an OPC-UA Server, start the server, and close the script without stopping it again.

    from opcua import Server
    
    url = "opc.tcp://127.0.0.1:50"
    server = Server()
    server.set_endpoint(url)
    server_name = "OPCUA_Simulation_Server"
    addspace = server.register_namespace(server_name)
    node = server.get_objects_node()
    Parameters = node.add_object(addspace, "Parameters")
    boilingHeat = Parameters.add_variable(addspace, "boilingHeat", 100e6)
    boilingHeat.set_writable()
    server.start()

Is there any way that get control over that OPC server so that the parameters (boilingHeat in my case) can be reused in another python script?

zerox1212, 2020-07-22:
I don't understand what you are asking. Are you new to Python? This looks like a basic Python question, not an OPC UA question.

Your other script has to import this one if you want to use ` boilingHeat` again.

sphemanth, 2020-07-23:
I am new to using OPC server-client mechanism. I am trying to create an OPC server and Client instance using the OPCUA package in python. My Client instance will write a variable and the Server will read and use it to perform some operations using the variable.

server.py(Psuedo code)

```
from opcua import Server

url = "opc.tcp://127.0.0.1:50"
server = Server()
server.set_endpoint(url)
server_name = "OPCUA_Simulation_Server"
addspace = server.register_namespace(server_name)
node = server.get_objects_node()
Parameters = node.add_object(addspace, "Parameters")
boilingHeat = Parameters.add_variable(addspace, "boilingHeat", 100e6)
boilingHeat.set_writable()
server.start()

t = 200
sa_t = 2.5 
for i in range(200/2.5):
    res = opera_function(boilingHeat.get_value(),i)

# Write result to a file and plot them

def opera_function(boilingHeat, iter):
    # operation using boilingHeat and iter
    return result
```

client.py(pseudo code)
```

    url = "opc.tcp://127.0.0.1:50"

    client = Client(url)

    client.connect()
    print("Client connected!")

    boilingHeat = client.get_node("ns=2;i=2")

    #h = 2.5 # sample time
    #tend = 200 # stop time
    n=int(tend/h)

    isClientConnected.set_value(True)
    t1_start = time.perf_counter()
    for i in range(n):
        t1_stop = time.perf_counter()
        elapsed_time = t1_stop - t1_start
        boilingHeat_in = ramp(elapsed_time)
        boilingHeat.set_value(boilingHeat_in)
        if elapsed_time < h*(i+1):
            print("Holding for some time:", h*(i+1) - elapsed_time)
            print("Current value for boilingHeat is : ", boilingHeat_in)
            time.sleep(h*(i+1) - elapsed_time)
        
    client.disconnect()
```
One limitation currently I am facing that my application performs execution in a temp folder and write to the folder only at the end of execution. The client instance keeps on sending the variable for say 200 sec and exit.

Here I want to access the intermediate result.

I tried to exit from the execution of script and writing the data to file and continue it again. But I do not have any idea of reusing/getting control over the running server instance.

Could you help me with this?

zerox1212, 2020-07-23:
You just want to handle the data the client sends to the server?

Just subscribe to your variable in the server and you will get notified every time the value changes, like of client writes a new value. See the example for subscription.

sa-chi, 2020-07-29:
A similar doubt for me as well. I don't know whether it is appropriate or not. 

Some times what happens is my python scripts exits without stopping the server if there are any exceptions (I didn't use any try except clause). This implies that my server is keep on running on that particular port in mysystem continuously without stopping. How could i get control over it in a new python script. Means something like
newscript.py
```
running_server = Server(port='<portnumber>') #<----- Should get control over server running on port
var = get_node("<node_id") #<------ which gives back the variable in the add_space

# perform some manipulation after getting the value
running_server.stop() #<------- Should stop the running server 
```
Is it possible to do like this? if yes,Could anyone help me with this?